cedricp/ddt4all #489: Megane 3 Installing second-hand stereo

johnroberts2k, 2020-11-09:
I had aftermarket stereo in my Megane. I wanted to replace with original so I bought one from breakers yard. After installing it doesn't turn on at all. Could it be immobilised? Is there a way to programme it to my car without needing radio code etc?

Thanks

luissilva1971, 2020-11-09:
Hi. I am also triyng to replace mine. Here is what i have read: Radio code: rhere are lots of renault code calculators free. google it, that should not be a problem. Turn on or off: do you achieve to connect to radio config through CLIP, DDT4all with modified ELM327 (CAN2) or other software? You need probably to change radio config to be compatible with your car (display, for instance)...  on the radio maybe you need to change display type for the one you have in your car (standard, Tomtom, R-link1, ...) and on the display config you may need to change config of radio type. That needs to be performed or through CLIP DDT4all, PyREN or Renolink...

luissilva1971, 2021-12-12:
Anyone with more accurate knowlegde can explain how it can be done on a megane mk3 2010.? i got the radio from a car without GPS, and i want to replace myno bluetoth stereo from the one i bought with bluetooh. Radio is already configurated for GPS, but radio on turn on. I read that it is necessary to change on navigation the radio type from R1 to R2 (keeping all other parameters) and perform some sort of reset!!!! PLEASE HELP. I wanted so much to do this for so long...  but i am afraid i brick my tomtom carminat. I would be forever grateful. Best regards.

luissilva1971, 2022-05-23:
I understand less and less. When changing the radio from R1_08 to R2_08 in the navigation module NFA_v0. 6, i typed write conf (expert mode). After i turn the radio off, closed car. Then i insert on car key again  
The message "system not configured" appeared on the screen. No request for radio code, nothing appear more on the screen. 
I went / am in panic stage. 
I went back changing to R1_08 radio type, pressed write conf, turn off radio and removed car key /close car . Removed R2_08 radio and placed again my R1_08 original radio again om its place. Turn it on and the GPS started working again (no message saying "system not configured...", but ... the radio bar / outdoor temp is not showing. The radio is playing, but the station's info bar is not showing, whem i press volume up, or even outdoor temp, nothing is showing on screen. In DDT4all, the R1_08 radio still appears in NFA v0.6, which is the one that is fitted. If I didn't change anything on the radio and in NFA v0.6 it was R1_08 why the bar is not showning? 
Is there any way i went back to old config? I have pyren and old ddt dump file besides ddt4all. Can you explain step by step what should i do? A least to return to the situation i had before i change the radio. PLEASE!!!!!

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif