cedricp/ddt4all #490: DDT4ALL not detecting Stereo ECU

johnroberts2k, 2020-11-10:
I had an aftermarket stereo/radio on my megane 3 which i've now replaced with the original one from breakers yard. However when I do automatic scan with ddt4all it doesn't detect any audio ECU. However, Renault CAN CLIP can see it and I can read the parameters from Renault CAN CLIP. But DDT4ALL can't find it.
Here is a screenshot of the ECU's DDT4ALL is only finding: https://i.imgur.com/R55n68l.png

And here is a screenshot of Renault CAN CLIP showing the Radio ECU: https://i.imgur.com/8P6jtkf.png

Can anyone help please. If I manually select it, will it work? which one do I manually select?

KarelSvo, 2020-11-10:
You must to connect with 12 and 13 instead of 6 and 14.
https://www.drive2.ru/l/468057702971474426/

johnroberts2k, 2020-11-10:
Thank you, I remapped the ELM to pins 12 and 13 and now it can detect Audio but my Audio ECU database cannot be found. I get the following error message:  https://i.imgur.com/2vVE6X3.png

KarelSvo, 2020-11-11:
Manually select R2_08v3_81.

johnroberts2k, 2020-11-11:
Thank you. I will try that. What is the MIU ECU that it also found on the multimedia network? I think this is the clock/temperature display unit.

KarelSvo, 2020-11-11:
The MIU is a data converter between Vehicle and Media Can Bus.

dexter2111, 2020-12-05:
Hi,
I have the same issue with my Renault Capture 2019, the firmware of the Ecu Audio is new ,that the DDT4ALL supported.
https://imgur.com/a/qJve06J
I don't know what to choose for the list , if i choose  Capture ph2  i get Navigation MFD v4.3.
One more question, can I modify the speed warning for the seat belt ?
I use a modified ELM for low/high CAN.

KarelSvo, 2020-12-06:
> I have the same issue with my Renault Capture 2019, the firmware of the Ecu Audio is new ,that the DDT4ALL supported.
> https://imgur.com/a/qJve06J
> I don't know what to choose for the list , if i choose Capture ph2 i get Navigation MFD v4.3.

Your VIN says Captur X87 with Radio Medianav RA43D.
ddt4all have found ULC4.
everything OK

dexter2111, 2020-12-09:
HI KarelSvo ,
Thank you for your replay.
After making the changes with DDT4ALL, do I need to restart the audio system?
Can you explain how?
I changed the color from Renault to Dacia, and the default language.
I did a factory reset of the audio system, but the changes do not become visible.
Can I  start the engine programable , if i enable RES from ECU(audio) , and from T4_HFM_DDT2000_SW14.45/REMOTE ENGINE STRAT/ Configuration ?
My remote don't have a  start engine button , he has a light button.

Thank you.

KarelSvo, 2020-12-12:
The Media Nav needs a hardreset to save the change.

luissilva1971, 2020-12-22:
Hi, guys. Have trying to replace a R108 original radio from my megane 3 van (also with a tomtom GPs device no live sersion) with a R208 radio from scrap yard (but that came from a car with no GPS, it has a standard LCD). With R208 radio fitted on the car I can change radio parameters, but I can not access to the navigation ECU, so i can change its settings from a R108 radio to a R208 radio (the icon on the left of Clip is greyed / no selection is possible). The strange thing is that with R108 fitted on the  car, the same icon it not greyed, it can be selected but no navigation parameters appears; also the original R108 radio dos not have the navigation parameters selected (it shows "NAV0/CDC0/AUX0" on the radio parameters) but navigation works fine with that radio. Can the settings be performed through DDT4all and ELM327 with the changed pins for multimedia - 6-14 to 12-13)?
Or the only ay i going to Renault car  workshop with "official" CLIP? Thanks for all the help and regards.

sametduyan, 2022-01-13:
this is how i solved it
https://www.drive2.ru/l/468057702971474426/

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif