cedricp/ddt4all #492: Anyone know what CL_CountryCfg does in DDT4ALL

AndyIreland, 2020-11-12:
hi - Anyone know what CL_CountryCfg does in the climate control section for X10 RENAULT ZOE in  DDT4ALL ? 

you have 'not configured'
then 'Zone 1(mild)'- which mine is set to default (I live in Ireland)
then 'Zone 2' 
       ' Zone 3'
       ' Zone 4'

is 'Zone 2' for colder countries than Zone 1 ?  --- or warmer countries than 'zone 1' 

And what does it actually change on the climate control settings of the car?

Thanks.

[image]

AndyIreland, 2020-11-12:
ah, found some info about this on google:

In the right window we find the parameter "CL_CountryCfg"
Change the values ​​(1-hot climate with smooth temperature change (default); 2-hot climate with sharp temperature changes; 3-temperate climate; 4-cold climate)
Enable Expert mode (in the menu from the top, the second icon from the left).
Save with the button "CL_CountryCfg", which is located to the right of the zone number.

https://xn--80aal0a.xn--80asehdb/do-my-self/tuning/tuning-lada-vesta/4844-kak-izmenit-nastrojku-klimat-kontrolja-v-zavisimosti-ot-klimaticheskoj-zony-na-lada.html

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif