cedricp/ddt4all #497: SSPP VelSatis

bobipln, 2020-11-24:
Hello. I try to write wheel sensor with PyRen in DDT mode, but command not work. I'm in ExpertMode, I write the code of sensor and put the "Mise a jour" button.
[image]
[image]
[image]

shrlnm, 2020-11-25:
It is looks like you are using outdated code. Try to update by pressing "Update" button in launcher.

bobipln, 2020-12-01:
I was able to enter the codes from the box below, showing which code applies to which wheel. But I have a problem with the visualization of the dashboard display. Whatever code I enter for the front right wheel, it does not want to show pressure and after a while it makes an error with a missing sensor. My question is where to look for the problem?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif