cedricp/ddt4all #498: PyRen

org07, 2020-11-25:
Is there an updated version of PyRen other than this link? Where can I download it packaged if available?
https://www.drive2.ru/b/501493851572142178/

org07, 2020-11-25:
Where can I download the current version of PyRen, don't anyone even know?

org07, 2020-11-25:
Où puis-je télécharger la version actuelle de PyRen, personne ne le sait?

org07, 2020-11-25:
¿Dónde puedo descargar la versión actual de PyRen, nadie lo sabe?

org07, 2020-11-25:
Где скачать актуальную версию PyRen, никто даже не знает?

org07, 2020-11-25:
Wo kann ich die aktuelle Version von PyRen herunterladen? Weiß es noch niemand?

shrlnm, 2020-11-25:
https://gitlab.com/py_ren/pyren/-/tree/master

shrlnm, 2020-11-25:
In current versions there is a button "update" in a launcher, it actually updates all files from gitlab.

org07, 2020-11-25:
> https://gitlab.com/py_ren/pyren/-/tree/master

Thank you for your answer, but here only, (PyRen-Master) has it loaded into (pyrene), it scans the ECUs in the car and finds them all, but when it comes to the end it gives the error I uploaded the photo. Packaged (PyRen), not available?

[image]

shrlnm, 2020-11-26:
These are not errors of software. It is statistic of ELM errors during last session. On your screen all counters are zeroes so it means that your ELM worked very well with ECUs. Packaged version has no advantages as so it have to be unpackaged to *.py files before usage.

org07, 2020-11-26:
> Bunlar yazılım hataları değildir. Son seanstaki ELM hatalarının istatistiğidir. Ekranınızdaki tüm sayaçlar sıfırdır, bu nedenle ELM'nizin ECU'larla çok iyi çalıştığı anlamına gelir. Paketlenmiş sürümün hiçbir avantajı yoktur, bu nedenle kullanımdan önce * .py dosyalarına paketten çıkarılması gerekir.
 shrlnm thank you. However this. https://www.drive2.ru/b/501493851572142178/  in this link

I am using PyRen184-installer.exe I have no problems. But when I install pyren-master it is having problems. I couldn't find the reason

shrlnm, 2020-11-26:
@org07 please explain what type of problem do you experience?

org07, 2020-11-26:
> @ org07 lütfen ne tür bir sorun yaşadığınızı açıklayın?

I will shoot the video tomorrow

org07, 2020-11-26:
> @ org07 lütfen ne tür bir sorun yaşadığınızı açıklayın?

I'll make a video of the problem tomorrow, I'll share it here

KarelSvo, 2020-11-28:
> 
> I am using PyRen184-installer.exe I have no problems. But when I install pyren-master it is having problems. I couldn't find the reason

how did you install? new files thrown in old program folder?

org07, 2020-11-28:
> > PyRen184-installer.exe kullanıyorum, hiçbir sorunum yok. Ama pyren-master'ı kurduğumda sorun yaşıyor. Sebebini bulamadım
> 
> nasıl kurdun eski program klasörüne yeni dosyalar mı atılır?

Yes, the files inside (pyren-master). I threw it into the old program that was installed.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836 yhis is ddt4all normaly

closed 2023-06-02 by Furtif