cedricp/ddt4all #500: Bricked head unit

Iogkadj, 2020-12-02:
Hi.  Has anyone tried opening up the lanr14 or 16 unit and connecting directly to the eMMC controller to try and write the firmware files to the flash and thereby restoring the unit?  (There is some documentation online for hacking eMMC chips for data dumping or writing)

I have a failed unit after failure to upgrade from 2.2.x.x to 3.3.x.x
Unit seems unresponsive,  unable to see the radnav ECU in dd4all and the usbs are dead.  From what i can gather the system erases the current firmware from the flash before writing the new one into flash.  And from the log files i can see that for some reason it failed when starting to read the new firmware from the usb and then terminated the process by writing the log files.   Now the unit is bricked.  I am looking into what options there are for recovering it.

AlMeu, 2021-04-28:
I have mo too some units bricked and so far I can't find a way to clean the data for write another time correct.
Do you have some more info?

gilnis2, 2022-12-26:
> Hi. Has anyone tried opening up the lanr14 or 16 unit and connecting directly to the eMMC controller to try and write the firmware files to the flash and thereby restoring the unit? (There is some [...]

Did you able to do it eventually? I have the same issue

AlMeu, 2022-12-26:
@gilnis2 
So far, what I know, what is dead in upgrade remain dead.
The simple way is to replace the unit.
If need some units unlocked for install in other car, leave an email for contact and I can help.

Iogkadj, 2022-12-26:
No I baught a new lanr16 unit and replaced the old one.   I have however recieved information since then that it is possible to put the device into recovery mode and have it read in new data from the usb.   I cannot remember the exact sequence any more.   I would have todo some digging arpund again for the correct info.    But the procedure involves pressing the two volume buttuns on the audio control buttons on the side of the steering wheel and the home button on the panel by the audio screen in the dash board.   There is a specific combination of button presses that should let you into the recovery mode once the device tries to bootup when u start the car.    I have not tested this yet but its on my schedule of todos.

Iogkadj, 2022-12-26:
Otherwise there is a group on FB called au bon coin du r-link.  One of the admins there can get you a brand new device ready for installation fora much much lower price than the official dealers.   Thats were i got my replacement for my car.

AlMeu, 2022-12-26:
In FB as well it is a group actualizari r-link where you can get a new or old unit.

Iogkadj, 2022-12-26:
Here is what i got on the recovery procedeure.   All i ask in return is that you Let us know if you get it to work.  I.e.  post the results here with us.

Procedure for recovery mode: 

Make a folder in the root of a usb drive called R-Link 
Put the relevant bios update file in the folder. (You may need a special recovery image file) 

1.   Start the engine of the car 

2.   Push the vol+, vol- on the remote behind the steering wheel and the power button on the R-Link,  press all three of them together a few times until the R-Link display comes on and you will be an presented with an option for emergency download. 

3.  Plug the usb stick into the usb slot and wait for R-Link to install the firmware file from the usb and restart a few times 

When this is working and after the update is done the R-link will ask you if you want to delete the update files on the usb stick.   Say no.   After that you can safely remove the usb stick.

AlMeu, 2022-12-26:
@Iogkadj 
> Make a folder in the root of a usb drive called R-Link Put the relevant bios update file in the folder. (You may need a special recovery image file)

What is "relevant bios update file"?
What is special recovery image file?

gilnis2, 2022-12-26:
Thank you all for the replies.

@Iogkadj 
I tried before to enter into this emergency download mode but wasn't able to.
I tried pressing the Volume +, Volume - (behind the steering wheel), and the On/Off button together for 40 seconds without success.
From what you say, maybe I should try pressing them together not for a long press but for small presses multiple times? I will give it a try tomorrow and will update here.
Thank you!

gilnis2, 2022-12-27:
@Iogkadj 
I tried now also to press short presses multiple times, no luck entering into emergency download mode :(
Thank you anyway

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif