cedricp/ddt4all #503: rlink 2 antitheft fix

Vuk031, 2020-12-10:
I have Renault Scenic 4
On device r link 2 on screen i have antitheft.
I virgined device with ddt, after that r link changed the vin number ( same as car vin) but i have same problem with antitheft on screen.
can anyone fix it?

KarelSvo, 2020-12-12:
How is someone supposed to help without seeing your ddt4all log files?

Timmytubby, 2021-04-19:
> I have Renault Scenic 4
> 
> On device r link 2 on screen i have antitheft.
> 
> I virgined device with ddt, after that r link changed the vin number ( same as car vin) but i have same problem with antitheft on screen.
> 
> can anyone fix it?

How do you Virgin it? Can figure out the procedure.

AlMeu, 2021-04-28:
@Vuk031 
I can remove the antitheft code but I need the unit on my desk.

Pekora78, 2022-01-21:
Buenas tardes. tengo 2  r-link 2  con el antirrobo activado ¿alguien tiene alguna solución?

AlMeu, 2022-01-22:
@Pekora78 
An email and maybe I can help you.

Pekora78, 2022-01-22:
daviciyo_78@hotmail.com este es mi correo y estaría muy agradecido por tu ayuda.

florin2025, 2023-04-16:
Hello.
can you send an email, maybe you can help me with an antithelf code.
thank you

AlMeu, 2023-04-16:
> @Vuk031 I can remove the antitheft code but I need the unit on my desk.

@florin2025 leave your email.

florin2025, 2023-04-17:
florin_2013@yahoo.com thanks

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

Arnaudkadjar, 2023-06-19:
Can someone help me with anti theth locked in R link Renault Kadjar. I ve tried to change VIN with renolink but no way, it doesn t change.thank

AlMeu, 2023-06-19:
@Arnaudkadjar leave an email and perhaps I can help you.

AlMeu, 2023-06-19:
@Arnaudkadjar you have to login on site for leave an email.

Arnaudkadjar, 2023-06-19:
My email adress 
Arnaud.gugnot@gmail.com

Zener-hub, 2024-01-15:
@AlMeu I have replaced my R-link two from another car after original unit got bricked. Changed the VIN on new unit before install but still antitheft activated. Can you help me removing antitheft? Thx

AlMeu, 2024-01-15:
@Zener-hub leave an email and we can talk.

Zener-hub, 2024-01-15:
Great, chandithp@yahoo.com
Thanks

lklik, 2024-03-29:
Hi AIMeu i have the same problem this is my email ; azizouaankour@yahoo.fr

gmzr0, 2024-05-09:
> @Zener-hub leave an email and we can talk.

is there any chance u could help me?

Goran-Barznje, 2024-06-08:
Hi, someone help for same issues after replaced the nav radio for Rlink2.
goran_barznje@yahoo.com

JohnnyX99, 2024-11-24:
Hello someone figured out how to virgin and deactivate theft mode ?