cedricp/ddt4all #504: ddt4all install on Linux/MacOS

LOZERE, 2020-12-11:
When typing the python ddt4all.py command in the folder wher eI have all the dd4all files, I have the following 

messageTraceback (most recent call last):
  File "ddt4all.py", line 8, in <module>
    from importlib.machinery import SourceFileLoader
ImportError: No module named machinery

Do I miss something ?
Thanks for any insight.

KarelSvo, 2020-12-12:
ddt4all is no longer compatible with Python 2
You have to write in the terminal python3 ddt4all.py

LOZERE, 2020-12-12:
I just have updated to python3, but when checking with python --version , I
still have the version 2.7.0....

Then when I am running the command python3 ddt4all.py in the ddt4all
directory I have the following:

Last login: Sat Dec 12 18:11:13 on ttys000

Marcs-MacBook-Pro-2:ddt4all-master {my name}$ python3 ddt4all.py

Traceback (most recent call last):

  File "/Users/{my name}/Desktop/ddt4all/ddt4all-master/ddt4all.py", line
10, in <module>

    import PyQt5.QtGui as gui

ModuleNotFoundError: No module named 'PyQt5'

Marcs-MacBook-Pro-2:ddt4all-master {my name}$

On another Mac I have installed the PyQt5 module , but then the program is
asking for another module and so on ...

So I don't what's wrong here

Thanks for any help

KarelSvo, 2020-12-12:
python3 -m pip install PyQt5

LOZERE, 2020-12-13:
Thanks for the answer, now after that module installation i have the
followng:

$ python3 ddt4all.py

Traceback (most recent call last):

  File "/Users//Desktop/ddt4all/ddt4all-master/ddt4all.py", line 14, in
<module>

    import PyQt5.QtWebEngine as webkit

ModuleNotFoundError: No module named 'PyQt5.QtWebEngine'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File "/Users//Desktop/ddt4all/ddt4all-master/ddt4all.py", line 17, in
<module>

    import PyQt5.QtWebKit as webkit

ModuleNotFoundError: No module named 'PyQt5.QtWebKit'

makayk, 2020-12-13:
> Thanks for the answer, now after that module installation i have the followng: $ python3 ddt4all.py Traceback (most recent call last): File "/Users//Desktop/ddt4all/ddt4all-master/ddt4all.py", line [...]

same issue here....

cristianbriscu, 2020-12-23:
pip install PyQtWebEngine
use this and comment out line 13, 16, 17, 18

for me worked, i have MBP M1 running python3.9.1 and brew under rosetta 2 

got the info from here: 
https://askubuntu.com/questions/763612/importerror-no-module-named-pyqt5-qtwebenginewidgets

berdyshev, 2021-05-13:
@cristianbriscu I have MBP 2019, python 3.9.5, installed PyQtWebEngine, commented lines you mentioned, but it still fails

```
Traceback (most recent call last):
  File "/Users/berdyshev/ddt4all/ddt4all.py", line 14, in <module>
    import PyQt5.QtWebEngine as webkit
ImportError: dlopen(/usr/local/lib/python3.9/site-packages/PyQt5/QtWebEngine.abi3.so, 2): Library not loaded: @rpath/QtQuick.framework/Versions/5/QtQuick
  Referenced from: /usr/local/lib/python3.9/site-packages/PyQt5/QtWebEngine.abi3.so
  Reason: image not found
```

makayk, 2021-05-13:
Here is the solution:
https://github.com/cedricp/ddt4all/issues/320#issuecomment-680182687

maksat20021999, 2022-11-24:
Hi 
After installing PyQt5, PyQtWebEngine and after comments 13, 16, 17, 18 lines still gives an error

$ python ddt4all.py
qt.qpa.plugin: Could not find the Qt platform plugin "xcb" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

zsh: IOT instruction  python ddt4all.py

KarelSvo, 2022-11-25:
https://noknow.info/it/os/install_libxcb_from_source#sec2-1

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

Dan-cer, 2023-06-03:
I got it to work finally with help from an advanced forum member in dacianer.de forum. However, it is too complicated to tell the steps here.
ecu.zip has to be part of the app, and there are python3 issues.