cedricp/ddt4all #509: I am getting an error message on ECU tweaker.

eraysgt, 2020-12-19:
My Bluetooth elm327 device is connecting.  but I can't change anything in the program.

ession : 10C0
Bluetooth Yöneticisi Mesajı : Re-intializing ELM...
Bluetooth Yöneticisi Mesajı : Intializing CAN protocol...
Olumsuz cevap : SubFunctionNotSupported (7F1012)
ELM Cevabı : 62210077050F0A16811F8080808032320F00000000 istek 222100
ELM Cevabı : 62210401011E1CC005210FF800 istek 222104
ELM Cevabı : 5003003201F4 istek 1003
ELM Cevabı : ERROR : 1002E2104010118
NO DATA
211CC005210FF800
NO DATA
 istek 2E21040101181CC005210FF800
Hatalı cevap : ERROR : 1002E2104010118
NO DATA
211CC005210FF800
NO DATA
 istek '2E21040101181CC005210FF800'
ELM Cevabı : 5001003201F4 istek 1001

eraysgt, 2020-12-28:
Friends, I'm sorry to say that the forum is dead

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif