cedricp/ddt4all #512: 2010 Laguna 3 TomTom Connection to the Navigation

sametduyan, 2020-12-24:
I cannot connect to the navigation in the car with ddt4all. I need to read the VINN and change it.
I tried all the options under DDT4ALL Navigation.

[image]

Can you help me ?

KarelSvo, 2020-12-24:
https://github.com/cedricp/ddt4all/issues/490

sametduyan, 2020-12-24:
> #490

Thanks for your support.
I can access other ECUs and read and write data.
Elm327 Usb Renault Pic18F25K80 Supported Ddt4I uses all compatible devices. Trouble accessing navigation ECUs

KarelSvo, 2020-12-24:
And did you use an adapter cable for the 2nd Can Bus?

sametduyan, 2020-12-25:
> And did you use an adapter cable for the 2nd Can Bus?

I could not understand exactly what you said. What should I do? Should I try it with a different elm327 device?
But with this device i can connect to other ecu and read and write data.

KarelSvo, 2020-12-25:
Then you ignored the link in my first answer.
There you can find out, what needs to be done in order, to connect to the separate can bus to radio.

luissilva1971, 2021-02-08:
Hello.
For a Megane 3 Grandtour 2010 with tomtom non-live version, which is the correct navigation UCC-ITM version? NFA v0.6? other.Thanks in advance.

bttarhan, 2022-01-13:
> I cannot connect to the navigation in the car with ddt4all. I need to read the VIN and change it. I tried all the options under DDT4ALL Navigation.
> 
> [image]
> 
> Can you help me ?

Hi.
I need to connect to navigation-ECU of my 2010 Laguna and change the VIN just as you need. Could you inform me whether it worked?

bttarhan, 2022-01-13:
I mean #490 or any other way. Did you solve the issue?

sametduyan, 2022-01-13:
> I mean #490 or any other way. Did you solve the issue?

this is how i solved it
https://www.drive2.ru/l/468057702971474426/

bttarhan, 2022-01-13:
Cheers.

bttarhan, 2022-03-05:
> > I mean #490 or any other way. Did you solve the issue?
> 
> this is how i solved it https://www.drive2.ru/l/468057702971474426/

It didn't work for me. ? Have I missed some point? I can spot several ECUs except for the Navigation. (No Navigation ECU is found despite the switch is toggled in either position.) What is the exact name for the Navigation ECU by the way?

AndyIreland, 2022-03-05:
> It didn't work for me. ? Have I missed some point? I can spot several ECUs except for the Navigation. (No Navigation ECU is found despite the switch is toggled in either position.) What is the exact [...]

If you cannot find automatically with the magnifying glass you will have to take a change and look in 'Navigation' and pick ecu manually from the list . anything with 'MFD' is more than likely gonna be a rlink 1 (evolution) touchscreen one 

if you see loads of red blocks on the parameters its highly likely you have chosen the wrong ECU for your car (or you need to increase CAN timeout to something like 600ms) 

[image]

luissilva1971, 2022-03-09:
Hello.
Which is the ECU navigation for a Carminat tomtom, non-live version?Thank you.

bttarhan, 2022-03-12:
> touchscreen

This didn't work either. I can detect several ECUs except for the navigation. (My nav device is Renault Laguna 2010 TomTom non-live version) When I toggle the switch I can detect nothing. Not even 1 ECU. I tried all these ECUs for the switch's both positions. Maybe, something's wrong with the connection. 

Any advice you could give is greatly appreciated.

AndyIreland, 2022-03-12:
> This didn't work either. I can detect several ECUs except for the navigation. (My nav device is Renault Laguna 2010 TomTom non-live version) When I toggle the switch I can detect nothing. Not even 1 [...]

OK , if magnifying glass not automatically detecting TomTom radio satnav then you will have to select MFD  manually . 

I can only talk for r-link1 evo with TomTom Live services because thats what I have fitted on my Renault ZOE EV. I would then manually select MFD v5.4 and change things like enable android auto and change other things on the r-link. Not sure what radio head unit you have on your laguna , not rlink 1 TomTom anyway I think they come out 2012 or 2013 ... carminat, is that the TomTom you have on yours?

Now there is another thing you can try and that is PyREN software and then select CAN2 and let it search for your TomTom (non live)

bttarhan, 2022-03-13:
> OK , if magnifying glass not automatically detecting TomTom radio satnav then you will have to select MFD manually .
>
> I can only talk for r-link1 evo with TomTom Live services because thats what I [...]

[image]

Carminat TomTom non-live is my nav unit. I also tried PyRen CAN2 as told, but returned 0 results.

bttarhan, 2022-03-13:
> [image]
[image]

> Carminat TomTom non-live is my nav unit. I also tried PyRen CAN2 as told, but returned 0 results.

[image]

> My OBD pins are soldered as different from above. Maybe that is why the ECUs are not found. I will change pins 3 and 11 with 12 and 13.

AndyIreland, 2022-03-13:
yeah - does sound at this stage wrongly wired ELM327

would my blog on wiring it up help:

https://andyfromsligo.wordpress.com/2021/07/18/wiring-up-an-obd2-splitter-cable-to-read-a-renault-r-link-radio/

bttarhan, 2022-03-15:
> yeah - does sound at this stage wrongly wired ELM327
> 
> would my blog on wiring it up help:
> 
> https://andyfromsligo.wordpress.com/2021/07/18/wiring-up-an-obd2-splitter-cable-to-read-a-renault-r-link-radio/

I solved it by soldering the connections as told... Later on, I found 2 ECUs. (Not for the navigation again) Therefore, I tried all ECUs in the DDT4All directory. After making some configurations, the nav system crashed. Now, I have a bigger problem. A message on the screen says "System not configured. Please connect diagnostic tool system". 

In summary; I've lost all displays and only have this message on my nav screen.

AndyIreland, 2022-03-15:
> > yeah - does sound at this stage wrongly wired ELM327
> > would my blog on wiring it up help:
> > https://andyfromsligo.wordpress.com/2021/07/18/wiring-up-an-obd2-splitter-cable-to-read-a-renault-r-link-radio/
> 
> I solved it by soldering the connections as told... Later on, I found 2 ECUs. (Not for the navigation again) Therefore, I tried all ECUs in the DDT4All directory. After making some configurations, the nav system crashed. Now, I have a bigger problem. A message on the screen says "System not configured. Please connect diagnostic tool system".
> 
> In summary; I've lost all displays and only have this message on my nav screen.
> 
> Any suggestions are welcomed :)

oh yes it sounds like you have pressed the wrong 'reset' button , maybe soft reset instead of 'hard reset' or pressed 'virgin' and virginised it  looks like maybe a trip to the garage to get the head unit re-programmed

KarelSvo, 2022-03-16:
Configuration reset means, reset all settings to factory delivery status. 
If you had previously opened all the ecu screens, you may be able to repair the ecu with the original data read. 
To do this, you have to search in the log files. 
So now make a Pyren dump. Modify this dump with the original data read from the earlier logs . Load the modified dump onto the ecu with Pyren.

bttarhan, 2022-03-16:
I used not only PyRen, but also DDT4ALL for changing the configs. In fact,
I tried to write new data under the Navigation ECU Menus in DDT4ALL. During
the writing process the power cut  off (I had not started the engine
before). This message "system not configured...." has been displayed on the
screen since I restarted the nav unit. The only thing I aimed for was to
change the status of center buttons to "active" again. (from "not fitted"
to "fitted").

KarelSvo ***@***.***>, 16 Mar 2022 Çar, 07:59 tarihinde şunu
yazdı:

AndyIreland, 2022-03-16:
@bttarhan - what MFD radio ECU shows up when you do a search on your car? with DDT4all or PyREN?

luissilva1971, 2022-04-03:
hello. could you explain step by step how you can restore settings from previous dump of a NFA v0.6 i made with pyren, please? i want to exchange radio type on mine (radio R1.08 to R2.08), and if something goes wrong i want to know how to restore! ... is it possible to change radio type using pyren or just with ddt4all? Regards

luissilva1971, 2022-04-30:
How to load previous pyren dump into ECU? Thanks

AndyIreland, 2022-05-01:
when you 'save' a dump file it should be saved by default 

C:\Program Files (x86)\PyREN\pyren\dumps

Then when you want to load it back into the Car's ECU go into 'rollback' and the dump text should be there 

[image]

luissilva1971, 2022-05-21:
Hi. Another doubt. When changing radio R1_08 to R2_08, i've changed the parameters on the audio, because the radio came from a non tomtom car. So i changed the NFA parameter on the radio to "present", and wrote the configuration. All ok. But in the NFA v0.6 (navigation)  I change the radio type from R1_08 to R2_08, with DDT4all in expert mode, and press "write conf", but when refreshing the screen, after writing, the type remains at R1_08, despite before refreshing, turn off the radio / navigation and turn it on again. What am I missing? To make the "hard reset" on Reset.all menu? Is it safe to do so (I don't run out of GPS)? I have some fears... but I would really like to change the radio. Note: when changing the parameter of the radio to have the NFA / GPS was just writing the parameter and it was saved in the radio, no need to reset. 
Another doubt: whenever I do the search, in CAN2 mode, 2 ECUs appear in Green (Radio R2_08 v3.81 and BICxxx something) and the navigation module - but in this case there is no direct correspondence, it is red, I manually select the NFA v0.6). Is it normal to be like this? Thanks

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif