cedricp/ddt4all #517: Bluetooth calling audio issue in R-link2 V9.0.34.611

Iogkadj, 2021-01-08:
Hi,  I am having problems with bluetooth call audio after installing a new R-link2 box with factory base version V9.0.34.611 
Everything seems to work fine,  my samsung s8 phone connects to r-link2 fine.  And I can play music and even use android auto.  But I dont get any audio from the telephone application.  no sound when key pressed when dialing in phone number through R-link2 and also if I make a call I don't hear any audio and the called person doesnt hear anything either.    Can anyone point me in the right direction to fix this? Is it possible by running a macro to reset the audio settings for the telephony application perhaps? 

I have a 2015 Kadjar with Bose audio system.

Iogkadj, 2021-01-08:
p.s. I have found the macro below for a problem that sounds similair to mine.    But I need information on how to run it.  Do I need DDT2000 to be able to run it? 

https://gitlab.com/py_ren/pyren/tree/master/pyren/macro/rlink2?fbclid=IwAR3IVOBQ2CHTyOXu9-qaEyf1upt2ky6WDvG-zl2fMCcDQWLaUSg05DHI59Y

shrlnm, 2021-01-08:
This macro can helps but it is for pyren
Before applying this macro  you need to downgrade to 7.

Iogkadj, 2021-01-08:
Hmmm.   Ok.   Thank you for your quick reply shrlnm.  Its very much appreciated.   Ím thinking as an alternative,  since this is a brand new device with factory installed firmware, would it be possible that the dealer didnt activate everything that needs to be activated?  Or,   is it possible to use pyren to manually enter upgraded commands to my new V9.0.34.611 for accomplishing the same as this macro script will do to a V7?
Reason I ask is i dont have good experiance from going between branch versions of the firmware as i already bricked one device already going from V2 to a V3 on a LANR14 R-Link2 device.  So Im thinking about all possible options.  ;)

igno77, 2021-01-11:
@shrlnm  do you have a macro for Acoustic on v9.0? This will help a lot of people with this problem. This macro will be different for Bose or standard audio system? I have an echo problem ONLY when I call with iphone connected by Carplay (the guy in the other side listen the own voice). This happen only in this condition

shrlnm, 2021-01-13:
I had the same problem on my device and did not find other way to solve except downgrading to v7, applying a macro and then upgrading to v9 again. It is looks like they've changed something in processing of this parameter in v9

ohayyam, 2021-01-13:
> I had the same problem on my device and did not find other way to solve except downgrading to v7, applying a macro and then upgrading to v9 again. It is looks like they've changed something in processing of this parameter in v9

@shrlnm as you know it happened to me too.. I downgrade to V7 applied your magic macro then upgrade to V9 and finally fixed the problem. 

I think it is not Rlink2 problem... it is problem of phone BT connection or ROM since when i change my android phone ( Note8 SM-N950F / DS ) rom from Android 9 custom rom to another android 10 custom rom problem started without touching the rlink2. 

Now i found another quick "solution" that when i answer phone call and during hearing robotic voice from other side just disconnect bluetooth and connect again on phone then voice is good. But each call you have to repeat procedure;  disconnect BT and connect again :))

As a result re installing android rom MAY fixed my problem on rlink 2 V9 version. if i re install to my older android 9 custom rom. i will inform here ...

For your kind informations

Iogkadj, 2021-01-13:
Hmm,  i have two phones i've tested,   galaxy s8 and galaxy s10.  I have the same problem on both.   I dont hear anything.   But the other party will hear robotic voice.

Iogkadj, 2021-01-13:
Shrlnm,  how did you downgrade from V9 to V7?  Is there an unofficial downgrade file floating around somewhere on the web or did you go to the Renault dealer for the downgrade?  Also i have bootloader version 5661 on my V9,  doesnt it require downgrading the bootloader aswell if i downgrade to V7?

Iogkadj, 2021-02-14:
I decided to just go for it and try out the rlink2_accoustic_8.cmd macro on my v9.0.34.611 boot version 5661 to try and solve the problem.   And it worked!!   Im glad i tried.  Seeing that the version on my device is the native version i was hesitant in downgrading.   Am so glad i didnt have to.   Now i have a fully functional v9

kaska9986, 2021-03-22:
Hola. Yo tuve el mismo problema y también probé a instalar el macro de la V8 para mi V9 y funciona. Arregle el audio de mi teléfono cuando está conectado al Rlink. Gracias

RamyarHP, 2021-03-23:
> I decided to just go for it and try out the rlink2_accoustic_8.cmd macro on my v9.0.34.611 boot version 5661 to try and solve the problem. And it worked!! Im glad i tried. Seeing that the version on [...]

Hello sir, I have updated from v3 to v9, and some of the functions now are not working, please can you tell me how did you do yours? I'm really not good at programing, I would be really thankful

Iogkadj, 2021-03-23:
RamyarHP,   what functions exactly are not working for you on your V9 R-Link2?    The solution discussed in this particular thread relates to audio not working in bluetooth phone calls.   It only fixes audio issues and bluetooth also.

RamyarHP, 2021-03-23:
> RamyarHP, what functions exactly are not working for you on your V9 R-Link2? The solution discussed in this particular thread relates to audio not working in bluetooth phone calls. It only fixes audio issues and bluetooth also.

To be honest mine mainly is when I go Reverse the camera wont show (though when I go to settings its working, it just doesn't automatically shows it with Reverse) and the automatic parking is also not working. mine was v3 and I just assumed I can update to v9, and now I was really sad and desperate that's why I commented in here, please im sorry if mine is not relevant.

Iogkadj, 2021-03-24:
I think you need to downgrade to the first version of  v2.2.x.x   and then upgrade again by going through all the versions leading up to v9 to retain the complete functionalities through the upgrades.     The audio macro will not help you with automatic parking and the camera,  only audio issues.   I would reccomend you join the facebbok group "Au bon coin du R-LINK"  and ask Adrien Borgognone to help you out.  He can walk you through the process

RamyarHP, 2021-03-24:
> I think you need to downgrade to the first version of v2.2.x.x and then upgrade again by going through all the versions leading up to v9 to retain the complete functionalities through the upgrades. The audio macro will not help you with automatic parking and the camera, only audio issues. I would recommend you join the facebbok group "Au bon coin du R-LINK" and ask Adrien Borgognone to help you out. He can walk you through the process

Thank you just so much sir, now everything has been solved except for that audio which is none existence at all, no radio or Bluetooth or mp3 (when you play the audio with flash drive or Bluetooth, it goes 1 or 2 seconds without sound of course and stop as it the output audio is not stable) and also the ambient light every 10 seconds or though it will flicker to another color. but thanks so much for the lead, I will contact that group and hopefully they will be able to help

RamyarHP, 2021-03-25:
> I think you need to downgrade to the first version of v2.2.x.x and then upgrade again by going through all the versions leading up to v9 to retain the complete functionalities through the upgrades. [...]

Hello again sir, I managed to fix the parking sensor and blind spot sensor too, but now only the sound doesn't work, there is no sound at all not in radio,BT or USB, please can you guide me to fix that as how you did yours, I would be really thankful, if you want I will drop my email too, thanks a lot (and i contacted that guy, he said I cant help you with your problem really)

Iogkadj, 2021-03-25:
You need to install python (i used python 2.7 and installed it to the root of my c drive) and install the pyren-master folder as a subfolder of c:\python27  then you need to run pyren via python.exe ( /pyren-mastef/pyren_launcher.py )
Once you have launched pyren you select the macro button in the bottom right corner of the pyren_launcher window and navigate to the macro file matching your r-link2 version  (it comes with pyren and is in one of the pyren-master subfolders) select it and run it.   Voila.  R-link2 Audio Problem fixed

RamyarHP, 2021-03-26:
Thank you a lot sir, you have really helped me a lot, I will go and get an OBD and will try it, I really cant wait to see it work, I will let you know, thanks again please.

RamyarHP, 2021-03-26:
> You need to install python (i used python 2.7 and installed it to the root of my c drive) and install the pyren-master folder as a subfolder of c:\python27 then you need to run pyren via python.exe [...]

I installed python 2.7 too and downloaded the pyren-master folder and put it in the python folder in C:\ and when I try to open pyren_launcher it just immediately closes after double-clicking on it, I don't know why its not opening?

Iogkadj, 2021-03-26:
You need to pipe it,  as in, you use python to run pyren.   You dont run python and then pyren.     
The launch string is something like the following example: 
 C:\pyren27\pyren.exe /pyren-master/pyren_launcher.py

igno77, 2021-03-26:
@Iogkadj This macro is perfect for fix the problem of no audio in the BT voice call but a lot of people have another problem that till this moment is unknown... When you use the phone with BT it's all right but when you connect the iphone with CarPlay and you made a phone call with it, the guy that is on the other side listen the echo of the own voice! This happen also in my Rlink born with 8.0 and now 9.0, do you have a solution? I've compared the achoustic and phone string from an RLink without this problem but it's the same of mine... Tested with more of 1 iphone, the same problem! Thanks

Iogkadj, 2021-03-26:
You can use this string,  just adapt the com port nr to the same com port you use for your elm327 device.
"c:\Python27\python.exe  _pyren_launcher.py -p COM6"

Iogkadj, 2021-03-26:
I think i heard somwhere that there is a known problem with latest iphones and rlink2 and also the latest version of android.   Its related to the os of the phones not rlink2.  Will be fixed witg updates from the phones vendors with time

Iogkadj, 2021-03-26:
Also iphone and carplay dont work on r-link unless your usb ports are usb3   so maybe you need to upgrade the usb ports of your car.   Mine also came with usb 2 as my car is model 2015.   I just baught the upgraded usb port module and replaced the old one.   
In addition to this there is a possibility that iphone carplay dont work unless you have the newer rlink2 head unit.   The old one is called LANR14 and the newer unit is called LANR16   the newer unit in combination with the upgraded usb3 ports works with iphone.

igno77, 2021-03-26:
@Iogkadj  yes yes, I know all and I'm ok! I have a native 8.0 RLink2 with boot 5608 and new usb port compatible with Carplay, I use only original usb cable and I have this echo problem from IOS 13 till last 14.4.1....it's not a problem of iPhone but it's a configuration inside RLink acoustic. I've tested my iphone in a different Megane with the same RLink (version and boot) and it works perfectly so it's not linked to the phone.

Iogkadj, 2021-03-26:
Igno77.  From what ive read the problem lies within the ios os as iphones with ios lower than v13  work fine on r-link2 devices that have problems with ios v13+    some ios updates have also been issued by apple to deal with the problem as it affects more devices than just R-link2.  Infact if you google the same sort of problem with head units from other manufacturers they also have the same problem with iphone and carplay.   Also android or samsung (cannot remember which one) has also issued a statement for a similair problem with the latest android os.  Which version was the other head unit you tested this in?  Was it running the same r-link2 version and boot  loader?   

https://www.google.com/amp/s/www.gps-rlink.com/2019/09/25/r-link-2-bluetooth-problem-with-apple-ios13/amp/

https://developer.apple.com/forums/thread/123991

https://www.google.com/amp/s/appletoolbox.com/carplay-not-working-after-ios-update-how-to-fix/amp/

https://support.apple.com/en-us/HT210892

https://www.autoevolution.com/news/the-latest-iphone-update-is-causing-new-problems-on-carplay-148902.html

https://lifehacker.com/what-to-do-if-you-cant-make-android-auto-calls-1845968429

Iogkadj, 2021-03-26:
You have to have the LANR16 unit for apple carplay to work on R-Link2 though.   But android auto works on both LANR14 and LANR16

Iogkadj, 2021-03-26:
You can have the same r-link2  software on either lanr14 or lanr16   but only lanr16 has the correct hardware for apple carplay to work

Iogkadj, 2021-03-26:
In anycase the audio macros with pyren  resets all the audio parameters with all the correct values.   If you have audio problems with r-link2 after using the macros i would be surprised if it turns out to be due to rlink2.

Iogkadj, 2021-03-26:
But if you have r-link2 with native v8  then you probably have the lanr16 device

Iogkadj, 2021-03-26:
https://www.autoevolution.com/news/good-luck-explaining-this-mysterious-carplay-bug-hitting-lots-of-iphones-150270.html

Iogkadj, 2021-03-26:
Im posting these links just to show you that the iphone updates have been buggy lately when it comes to carplay.

Iogkadj, 2021-03-26:
P.s.   why are you still on v8 for rlink2??  I would upgrade to the latest version of v9 and update the phones software also and see if it works.   There are some improvements made in v9 when it comes to audio,  bluetooth and android auto/carplay

Manu-El83, 2022-01-28:
@logkadj hello could you help me please? I have exactly the same problem on my megane 4 GT and despite the macro 8 nothing changes... I have the r link version and the same boot... could you just tell me your radio reference? Mine is 281155532R

Manu-El83, 2022-01-28:
> Hi, I am having problems with bluetooth call audio after installing a new R-link2 box with factory base version V9.0.34.611 Everything seems to work fine, my samsung s8 phone connects to r-link2 [...]
@logkadj hello could you help me please? I have exactly the same problem on my megane 4 GT 2016 and despite the macro 8 nothing changes... I have the r link version and the same boot... could you just tell me your radio reference? Mine is 281155532R

Manu-El83, 2022-01-28:
@Iogkadj thank you

Manu-El83, 2022-01-28:
@Iogkadj hello could you help me please? I have exactly the same problem on my megane 4 GT 2016 and despite the macro 8 nothing changes... I have the r link version and the same boot... could you just tell me your radio reference? Mine is 281155532

Iogkadj, 2022-01-28:
It should just work if do e correctly.   Try a hard reboot of the Rlink2 device by holding the power button for about 15 secs or until you see the rlink2 logo.   See if it works after that.   You could also try the macro for version 7.

Manu-El83, 2022-01-28:
Thanks but no -(

Iogkadj, 2022-01-28:
The partnmuber for the rlink2 device doesnt matter because its the same software for all rlink2 devices.   There isnt a special version for megane and another for clio.   They are all the same rlink2 software just with different revision numbers.  So if the macros dont solve your problem then its not a problem with the devices bluetooth audion.  Its something else.   Have you tried a different phone?   Its possible you have a phone thats not compatible.   What kindof phone do you have?  Is it an iphone??  And what happened when you ran the macro?   What did the program output after running the macro?

Iogkadj, 2022-01-28:
The unique partnumber for my rlink2 device is just unique to the fact that its for a car with bose system.   I have a kadjar bose edition.   But the software is the same as for cars without bose.

Manu-El83, 2022-01-28:
I have an iPhone 11 Pro Max in iOS 15. It restarted the r link normally and then I did a 15-second reboot of the rlink but unchanged. Once the macro installed pyren s closed. I tried with a Samsung and it works well

Manu-El83, 2022-01-29:
@logkadj Or do you have macro 8? Would you have a link please?

Iogkadj, 2022-01-29:
The macros are included with pyren.   In a macro subfolder of pyren.    I do however think you have a different problem that this macro is intended to resolve.    You probably have have a problem with the compatibility of your iphone to the rlink2 system.   I think something needs to be updated on your phone.    Many people have had your problem and i think they have solved it but im not sure how as i dont have an iphone myself.  But ive heard of this.   It is specifuc to the latest models of iphone if i remember correctly.   Iphones above 11 i think.    You need to google for a different solution.    Try searching for problems with iphone and rlink2  or carplay and rlink2.   
The thing is if the system works perfectly with any phone (like a samsung phone) the system is then operating correctly.   Otherwise the problem would be the same with any phone.

Manu-El83, 2022-01-29:
Yes, I think there is a problem there too, but since we had exactly the same problem except you kadjar and me Megane and you solved the problem with macro 8 I thought maybe you had a pyren with a different version

igno77, 2022-01-29:
You are speaking about a lot of wrong things! There aren’t any problem with iPhone, Rlink and so on…this is a common problem when you use a virginized Rlink or not configurated system! If your Rlink was born with 7.0 and now is 9.0, you have to come back to 7.0 and run the macro with Pyren to solve the BT problem. After that you can go up to 9.0 without problem. Already done a lot of time with success

Manu-El83, 2022-01-29:
I believe v9 origin

Manu-El83, 2022-01-30:
@shrlnm 1 solution you?

Iogkadj, 2022-01-30:
https://www.google.com/amp/s/www.gps-rlink.com/2019/09/25/r-link-2-bluetooth-problem-with-apple-ios13/amp/

Iogkadj, 2022-01-30:
https://renault-connect.renault.com/forum/question/r-link-bluetooth-connection-issues/

Iogkadj, 2022-01-30:
https://www.renaultforums.co.uk/threads/megane-r-link-2-iphone-bluetooth-streaming-issue.524026/

Iogkadj, 2022-01-30:
https://discussions.apple.com/thread/8077573

But i agree with igno77,  it might be worth downgrading to v7 and try running the macro.    But for me the v8 macro worked straight away on my v9 box

Manu-El83, 2022-01-30:
@Iogkadj My problem is not Bluetooth rlink but by connecting to CarPlay

Iogkadj, 2022-01-30:
Ahh ok.   Then have a look at the links i posted for resolving your problem.  The macros are only for bluetooth audio problem.    

Just to recap,  you plug your iphone v? To the rlink2 via usb cable and you have problems with audio in phonecalls?   Or you have problems streaming music?

Manu-El83, 2022-01-30:
Audio phonealls only

ionitagelu, 2022-04-21:
> Ahh ok. Then have a look at the links i posted for resolving your problem. The macros are only for bluetooth audio problem.
> 
> Just to recap, you plug your iphone v? To the rlink2 via usb cable and you have problems with audio in phonecalls? Or you have problems streaming music?

Hello, any news about CarPlay USB cable issue. I'm experiencing a similar issue, iPhone connected to RLink 2 native v9 unit, and the microphone volume is too low in phone call, the other party hear me weak. If I'm using bluetooth connection between iPhone and RLink microphone volume (sensitivity) is good. Regards,

Manu-El83, 2022-04-21:
Hello, that's exactly it! When I plug my iPhone 11 Pro Max into the rlink 2, the person at the other end of the phone hears echo and hears me from afar. While I hear well. But if I switch with Bluetooth rlink it works well. I tested pyren macros but nothing does about it. We had not managed thanks to a person to take over the audio parameters of a similar Megane and create an identical macro but nothing either.

Emmanuel Molines

GioNzz, 2022-10-24:
Hello,
I am having at the same issue described by you. I have no audio during calls and the caller hears my robotic voice "robotic". 
My R-Link was native 2.2 (Boot 5615) and has been upgraded up to version 9.0.35.50x.
Would the script for version 8 be okay in my case?

Lauddt94, 2023-04-26:
Is there an update on @Manu-El83 @GioNzz how this issue was solved?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

[8 more replies]

closed 2023-06-02 by Furtif