cedricp/ddt4all #518: 7f2e7f hatası??

mirack61, 2021-01-09:
[image]
For Megane 4 drl and tail
Help

KarelSvo, 2021-01-09:
The explanation for the negative response is in front of the bracket.

pilotkalem, 2021-05-04:
TR:
7F verdiğin komuta olumsuz cevap
2E verdiğin komut (write data by identifier -  tanımlayıcı adresini kullanarak kontrol ünitesine veri yazmak)
7F olumsuz cevabın tanımlama kodu

bazı fonksiyonlar sadece belirli diagnostik oturumlarıyla kullanılabilir. diagnostik oturumu dediğimiz 10C0, 1003,1060 vb gibi olan kısım. diagnostik oturumu açmak veya bazı durumlarda o diagnostik oturumundaki bazı fonksiyonları kullanmak için (senin örnekte 2E write data by identifier) bir güvenlik onayı almak gerek. 2703 ile seed sorulur, gelen seedden bir key hesaplanır ve bu araca gönderilir, kontrol ünitesi bu keyi onaylarsa istediğin konfigurasyonu yazmana izin verir. hatanın sebebi özetle bu şekilde.

it is because you need a security access with right diagnostic session ( 10 c0, 10 03 or 10 60). for all sessions  different calculation  methods are needed to have the key from seed. Without gaining access to right session after calculating the key from seed, you cannot use 2E function, or you need to change the diagnostic session.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif