cedricp/ddt4all #519: How do you change 'setpoint' parameter in DDT4al or PYREN?

AndyIreland, 2021-01-10:
Hi, - I have a Renault ZOE 22kw 2013 EV and I have always thought that when the interior heater climate control is set to deflect to footwell that its very poor. The heater is good out of every other vent like when deflected to windscreen or face vents on the dash but footwell very poor and my feet and legs feel colder. 

So yesterday I put a OBD2 adaptor into the CAN socket and put the climate control to feet area only and then checked with PyREN and it gave me this : 
[image]

as you can see the footwell deflection is only 48% and it will not go any higher - if i want all the heat to divert to my feet (which I do in winter and cold weather) I will need to have that 90-100% - but then I see the motor setpoint is at 48% so thats got to be the reason its heat wont go anywhere near to 100% down by my feet!

So, now to the question , How do I get to the setpoint section of the motor for the flap? - have looked in DDT4all and PYREN and cannot find this setting at all - any help please?

KarelSvo, 2021-01-11:
```
X10-X98-X87-X52ClimBoxGen1-0000-04
   <Parameter name="P009" agcdRef="PR038" codeMR="PR038">
        <Label codetext="51143" defaultText="CONSIGNE MOTEUR DISTRIBUTION PIEDS"/>
        <Unit codetext="31" defaultText="%"/>
      </Parameter><Parameter id="P009"><Computation type="FLOAT32"><Value>_OH_FEETVENTACTRREQ </Value><MnemoList><Mnemo name="_OH_FEETVENTACTRREQ" /></MnemoList></Computation></Parameter>

 <Parameter name="P010" agcdRef="PR023" codeMR="PR023">
        <Label codetext="13684" defaultText="FOOTWELL - VENT DISTRIB FLAP POSITION"/>
        <Unit codetext="31" defaultText="%"/>
      </Parameter><Parameter id="P010"><Computation type="FLOAT32"><Value>_IH_FEETVENTACTRPOSITION </Value><MnemoList><Mnemo name="_IH_FEETVENTACTRPOSITION" /></MnemoList></Computation></Parameter>
```

Find the Mnemo name in the ddt2000 Ecu file. Then you can see whether there is a write command.

AndyIreland, 2021-01-19:
thank you for that - i am a bit confused - I opened up file:
X10_X98_X87_X52ClimBoxGen1_v08_3_20160722T152536.xml
and got to this part - do i change anything here @KarelSvo ?
[image]

Electrotom, 2021-01-21:
Hi Andy, I like your approach to zoe.  I would like to come to the function and improve the heating while charging let's help. @AndyIreland

AndyIreland, 2021-01-26:
when I go into PyREN it finds ECU 11119 X10-X98-X87-X52ClimBoxGen1_0000_04_A   ECU. 

But if I go into ECUs folder  of PyREN folder in Program Files (x86) on windows there is not a file of 11119 X10-X98-X87-X52ClimBoxGen1_0000_04_A.xml    there are only xml files of gen2ClimBox - gen3 ClimBox etc

Electrotom, 2021-01-26:
pyren now plan to run DDT4all I searched and this parameter is not there at all.  overall I hope there will be better pyrene.  I count about 2 days and I could have tried and let you know what to do with me.

AndyIreland, 2021-01-26:
> pyren now plan to run DDT4all I searched and this parameter is not there at all. overall I hope there will be better pyrene. I count about 2 days and I could have tried and let you know what to do with me.

sorry - but because of translation I dont know what question you are asking me @Electrotom  - I am not ignoring you honestly :)

AndyIreland, 2021-01-26:
@Electrotom  - if you are asking 'how can you keep your heating on inside your Zoe when charging?" - I think this is what you might mean ? - If so  its not easy on the Zoe because as you are charging it just blows out cold air out of the vents , there might be a workaround if you press the 'preheat' button on the Zoe keycard but thats the only way I think you can do it

AndyIreland, 2021-02-16:
> Find the Mnemo name in the ddt2000 Ecu file. Then you can see whether there is a write command.

if there is not a write command there then does that mean parameters cannot be changed at all then? - can we write in our own settings into the xml file?

flame00111, 2022-01-14:
Hi guys my inlet throttle valve always open its not closing. My car is stage 1 and its 115 hp and ı go chiptuning station and delete the stage 1 and now its stock 90 hp clio 2014 1.5 dci edc. And When i look ddt and pyren its setpoint 89.80. And its always 89.80 not closing. How i set it % 9 - 10 with ddt ? I reset it pyren its still 89.80. And my fuel consumption very high. My lambda not 1 and afr not 14.7. Thanks. Please help me. I see inlet throttle setpoint %95. And i how i setpoint to %9 or whatever stock ?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif