cedricp/ddt4all #522: Change VIN

p3DRu1, 2021-01-20:
@Shrlnm can u help me change VIN on one unit I buy for my car from another car?
Thank u in advance!
My email is: p3dru1@gmail.com 
I pay it if you need.
Megane 4

nightrun387, 2021-01-23:
@shrlnm 
I have the same problem. I upgrade to an R-link2 from a salvage car and now it's in anti-theft mode.
nightrun387@gmail.com
Ref. 281155986R 
Date: 2019/09/28
Can you help?
Thanks!

KarelSvo, 2021-01-23:
use pyren
save a dump with pyren mod_ddt
change the VIN in the dump
configuration reset of the radio
load dump with pyren mod_ddt

p3DRu1, 2021-01-23:
> use pyren
> salvar um dump com pyren mod_ddt
> alterar o VIN na
> configuração de dump reset do radio
> load dump com pyren mod_ddt

Can you tell me how to do it? A little step by step? I have pyren, ddt4all and a lot of computer experience. I must say that I don't know how I did it well but when connected with DDT4ALL I selected "HARD RESET" to try get free of bloked screen, does the procedure you are referring to work the same? And then I have to check all the settings one by one without knowing the correct parameters yet, but that will be another problem later. THANK YOU SO MUCH FOR YOUR RESPONSE!

nightrun387, 2021-01-24:
> use pyren
> save a dump with pyren mod_ddt
> change the VIN in the dump
> configuration reset of the radio
> load dump with pyren mod_ddt

this is the dump file created by pyren, how can i change the vin?
i´m using windows is there a specific mode for entering mod_ddt? or is using the normal pyren_launcher.py link?
thanks 
[1606707568_11850.txt](https://github.com/cedricp/ddt4all/files/5862853/1606707568_11850.txt)

KarelSvo, 2021-01-24:
22F18C:62 F1 8C 43 41 50 47 48 39 4B 31 58 32 36 38 38 33 36 30 30 32 32 20 FF FF FF FF 
Put 43 41 50 47 48 39 4B 31 58 32 36 38 38 33 36 30 30 32 32 20 in a hex to ASCII converter.

nightrun387, 2021-01-24:
I made the alterations for the vin but the radio still assumes the old one :(

AlMeu, 2021-04-28:
@nightrun387 
The VIN can't be overwriten.
I can clean the units by VIN and antitheft code, but I need to have the unit on my desk.

shrlnm, 2021-04-28:
VIN may be changed only after virginazing the unit. But for virginize antitheft locked unit you need to manipulate with pins on unit pinout, generally on the table.

AlMeu, 2021-04-28:
I agree.

msuhaibs, 2023-03-12:
Hello, In moving RLink2 from one car to another, besides VIN and Vehicle ID, is anything else needed changing such as HUID? Also how much it affects if the unit (LANR16) is taken from a different type of car e.g. a Megane IV RLink2 in a Kadjar or a Kadjar petrol unit in a Kadjar diesel?

shrlnm, 2023-03-13:
@msuhaibs VIN does matter only if you bought something in the r-link store. HUID do not  influence anything and you can't change it. The parameters dependent on car body are audio parameters, rear camera lines and some others dependent on car options.

msuhaibs, 2023-03-13:
@shrlnm. If the VIN is only for keeping old r-link store purchases active then is it just the vehicle ID that triggers anti-theft? In that case, if the unit is coming from a much newer car, say 2020 or even later, shall it not benefit from free maps etc.? Although audio parameters, etc. are based on car specification, but should the unit work in principle and can be fine tuned later on with ddt4all?

msuhaibs, 2023-03-13:
@shrlnm thanks for your clarification. Does the triggering of anti-theft (with lock symbol) taking place after putting the unit in a different car (mismatched acode) actually writes any extra non-editable file or fields making the unlocking more difficult? Also, as long as anti-theft is not triggered, is VIN an editable field via ddt4all?

shrlnm, 2023-03-14:
@msuhaibs New unit have acode=0. UCS module periodically sends constant vehiceID to the CAN bus. When you connect r-link2 to the CAN for a first time, it reads vehicleID from the CAN and save it in acode. Further it compare their acode with vehicleID from the CAN bus

if acode==0 then acode=CAN.vehicleID
if acode!=0 and CAN.vehicleID!=acode then anti-theft = blocked

msuhaibs, 2023-03-14:
@shrlnm thanks for your reply. In an earlier reply to me you wrote, "Regarding unit from other car the answer also yes. It may be done but it will be not simple job to configure it appropriately.". I wonder if and how one can save configuration from his own rlink2 or a very similar car (for example, 1.6 Kadjar diesel) and load it to the newer unit that came from a different derivative, Megane 4, for example?

msuhaibs, 2023-04-01:
Aside from anti-theft, what potential problems can we expect transferring an rlink2 head unit from a 2020 manual, petrol Kadjar in a 2016 diesel automatic Kadjar? Shall it be necessary to change the car configuration in ecu configuration via ddt4all? Does the change in vehicle ID and potentially VIN results in the loss of any already activated features and maps?

shrlnm, 2023-04-01:
@msuhaibs Hello. Read my article https://www.drive2.ru/b/643019689764615344/
Probably it will answer your questions.
R-Link2 may be transferred from one car to another. VIN influence only on your stuff in r-link store

msuhaibs, 2023-04-01:
@shrlnm thanks. I will have a look.

AlMeu, 2023-04-01:
> Aside from anti-theft, what potential problems can we expect transferring an rlink2 head unit from a 2020 manual, petrol Kadjar in a 2016 diesel automatic Kadjar? Shall it be necessary to change the [...]

Beside anti-theft, you need to full reconfigure the unit, because very less probability to match a unit from identical car.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif