cedricp/ddt4all #523: language

ntmbn, 2021-01-21:
how can i change the language? 
i choose english but it is french

yuksel967, 2021-01-21:
google translate

ntmbn <notifications@github.com>, 21 Oca 2021 Per, 23:34 tarihinde şunu
yazdı:

> how can i change the language?
> i choose english but it is french

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virüs
bulunmuyor. www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

cedricp, 2021-01-21:
The database is a mix of French and English, there's unfortunately nothing you can do (translate or ask here).

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif