cedricp/ddt4all #525: My question to dear Shrlnm

org07, 2021-01-29:
Mr. Shrlnm PyRen why do you close its advanced sections and limit its features?

We are looking for Mr. Zmiterm very much, she was clearly packaging the programs pyreni and ddt4all.

KarelSvo, 2021-01-29:
> We are looking for Mr. Zmiterm very much, she was clearly packaging the programs pyreni and ddt4all.

He didn't packaging these programs, he just made them known and distributed. 
Which advanced sections and limit its features do you mean? You speak in riddles.

org07, 2021-01-30:
> > Bay Zmiterm'i çok arıyoruz, açıkça pyreni ve ddt4all programlarını paketliyordu.
> 
> Bu programları paketlemedi, sadece onları tanıttı ve dağıttı.
> Hangi gelişmiş bölümleri ve özelliklerini sınırlandırmak istiyorsunuz? Bilmecelerle konuşuyorsun.

Ben sınırlandırmak istemiyorum. Programdan çok önemli işe yarayan, bölümler çıkartıldı, Artık PyRen'in çokda fazla işlevliği kalmadı.

org07, 2021-01-30:
[image]

org07, 2021-01-30:
the program is constantly going backwards, while it needs to improve constantly. Useful sections are removed from the program

org07, 2021-01-30:
> > Bay Zmiterm'i çok arıyoruz, açıkça pyreni ve ddt4all programlarını paketliyordu.
> 
> Bu programları paketlemedi, sadece onları tanıttı ve dağıttı.
Zmiterm did not pack the program, but why can't pyren be downloaded anywhere else from its site? There is no 2nd site to download pyren in its bundled form other than his site. Pyren only has pyren-master updates on their site that they broke.

Marianpol, 2021-01-30:
ECM tab is just hidden for safety reasons. You can access it by typing 'cmd' in main module menu (menu that is on your screen), submitting that command and refreshing a module menu by going back and forth to any tab.

> Pyren only has pyren-master updates on their site that they broke.

What is broken with pyren-master?

> the program is constantly going backwards, while it needs to improve constantly. Useful sections are removed from the program

What else beside hidden ECM tab is making latest PyRen worse?

org07, 2021-01-30:
> ECM sekmesi yalnızca güvenlik nedenleriyle gizlenmiştir. Ana modül menüsünde (ekranınızdaki menü) 'cmd' yazarak, bu komutu göndererek ve herhangi bir sekmeye ileri geri giderek bir modül menüsünü yenileyerek erişebilirsiniz.
> 
> > Pyren'in sitesinde sadece kırdıkları piren-ana güncellemeleri var.
> 
> Pyren-master ile ne bozuldu?
> 
> > program sürekli olarak iyileştirilmesi gerekirken sürekli geriye doğru gidiyor. Yararlı bölümler programdan kaldırılır
> 
> Gizli ECM sekmesinin yanında başka neler en yeni PyRen'i daha da kötüleştiriyor?

Apparently, Pyren will now be aut renolink. As I wrote before, the program is getting worse rather than improving. Not everyone is a computer software expert, not everyone has the skills to write code. The simpler the program, the more sought after Pyrende used to be.

org07, 2021-01-30:
PyRen Aut i guess it's happening

org07, 2021-01-30:
I guess PyRen is Aut now

org07, 2021-01-30:
PyRen Aut diğer Renault program IN, The situation shows that we will turn to new programs that are in There are new programs for renault with no restrictions. It is paid but at least you solve your problem unlimitedly. The fee is not too expensive, it is very small numbers.

KarelSvo, 2021-01-30:
> the program is constantly going backwards, while it needs to improve constantly. Useful sections are removed from the program

I see exactly the opposite.
Pyrene has been continuously developed and improved by Marianpol.
You can permanently display the ECM menu by changing the line opt_cmd = False to opt_cmd = True in the mod_globals.py file. 
There is no intersection between free pyrene and commercial Renolink, the is different concepts.

org07, 2021-01-30:
> > program sürekli olarak iyileştirilmesi gerekirken sürekli geriye doğru gidiyor. Yararlı bölümler programdan kaldırılır
> 
> Ben tam tersini görüyorum.
> Pyrene, Marianpol tarafından sürekli olarak geliştirildi ve iyileştirildi.
> Mod_globals.py dosyasında opt_cmd = False satırını opt_cmd = True olarak değiştirerek ECM menüsünü kalıcı olarak görüntüleyebilirsiniz.
> Serbest piren ile ticari Renolink arasında kesişme yoktur, farklı kavramlardır.
> Pyren ile dökümleri yedekleyebilir, karşılaştırabilir ve geri yükleyebilirsiniz. Renolink ile bu mümk

Can you change this file for me and send it as zip from here? In the mod_globals.py file, change the line opt_cmd = False to opt_cmd = True to permanently set the ECM menu.

org07, 2021-01-30:
KarelSvo Thank you very much, very much

Marianpol, 2021-01-30:
PyRen was designed to be a CLIP copycat and it nearly is. And it is free. I know, that command line interface (CLI) is not a preferred way to operate in the PyRen, but user interface (UI) would take a lot of time to be implemented and I would rather spend this time on removing bugs and adding other features.

Renolink is a different story. It is more like DDT4ALL with additional advanced features (UCH, EPS, AIRBAG cleaner, TDB mileage correction, TDB EEPROM/FLASH upload download and more). You are able to use this function because Renolink developers have spend their time and money to buy these modules to work on them and discover theirs secret keys (APV/PIN) and other backdoors to do such advanced things. 
PyRen is free so I would have to pay by my own money to do the same thing. I've got other things to spend my money on.

If you need that additional features just go and buy your copy of the Renolink. Pyren probably would never have such large library of advanced features.

org07, 2021-01-30:
> PyRen bir CLIP taklitçisi olarak tasarlandı ve neredeyse öyle. Ve ücretsizdir. Komut satırı arayüzünün (CLI) PyRen'de çalışmak için tercih edilen bir yol olmadığını biliyorum, ancak kullanıcı arayüzünün (UI) uygulanması çok zaman alacaktır ve bu zamanı hataları gidermek ve diğer özellikleri eklemek için harcamayı tercih ederim .
> 
> Renolink farklı bir hikaye. Ek gelişmiş özelliklere (UCH, EPS, AIRBAG temizleyici, TDB kilometre düzeltme, TDB EEPROM / FLASH yükleme indirme ve daha fazlası) sahip DDT4ALL'a daha çok benziyor. Bu işlevi kullanabilirsiniz çünkü Renolink geliştiricileri, bu modülleri üzerlerinde çalışmak ve gizli anahtarlarını (APV / PIN) ve bu tür gelişmiş şeyler yapmak için diğer arka kapıları keşfetmek için bu modülleri satın almak için zamanlarını ve paralarını harcadılar.
> PyRen ücretsizdir, bu yüzden aynı şeyi yapmak için kendi paramla ödeme yapmam gerekir. Paramı harcayacak başka işlerim var.
> 
> Bu ek özelliklere ihtiyacınız varsa gidip Renolink kopyanızı satın alın. Pyren muhtemelen hiçbir zaman bu kadar geniş gelişmiş özellikler kitaplığına sahip olmayacaktı.

I bought Renolink already the latest version is very advanced

org07, 2021-01-30:
There will be one more question for you, can I delete the injection brain immobilizer with ddt4all, which file do I need to manipulate for it?

KarelSvo, 2021-01-30:
> I bought Renolink already the latest version is very advanced

What have you changed with Renolink that cannot be done with other software? 

> There will be one more question for you, can I delete the injection brain immobilizer with ddt4all, which file do I need to manipulate for it?
What do you think, how many different ecu are installed at Renault and what the answer is if you do not write details about your ecu?

org07, 2021-01-30:
I don't want to be disrespectful to Mr. Cedric PAILLE by advertising renolinkin here

org07, 2021-01-30:
ddt4all is our eye, our baby

Marianpol, 2021-01-30:
> There will be one more question for you, can I delete the injection brain immobilizer with ddt4all, which file do I need to manipulate for it?

There is no option to do that. You can do it Renolink only if you have quite old Sirus32 ECU.
In most cases you have to download an injection module EEPROM data using BDM with e.g. KTAG, edit it properly and upload it to the EEPROM back again. For some ECUs you have to edit also a FLASH, but firstly you have to know what ECU sits in your car, then search in google "your_ecu_name immo off" for more information.

org07, 2021-01-30:
> > Size bir soru daha olacak, enjeksiyon beyin immobilizeri ddt4all ile silebilir miyim, bunun için hangi dosyayı değiştirmem gerekiyor?
> 
> Bunu yapma seçeneği yok. Bunu Renolink'i ancak oldukça eski Sirus32 ECU'nuz varsa yapabilirsiniz.
> Çoğu durumda, BDM'yi örneğin KTAG ile kullanarak bir enjeksiyon modülü EEPROM verisini indirmeniz, uygun şekilde düzenlemeniz ve tekrar EEPROM'a yüklemeniz gerekir. Bazı ECU'lar için ayrıca bir FLASH düzenlemeniz gerekir, ancak önce ECU'nun arabanızda hangi ECU'nun oturduğunu bilmeniz, ardından daha fazla bilgi için google'da "your_ecu_name immo off" içinde arama yapmanız gerekir.

Dear Marianpol, you provided important information, thank you very much.

closed 2021-01-30 by org07