cedricp/ddt4all #530: It's clear that nobody can answer that

org07, 2021-02-03:
As with Can Clip, is it possible to access the car's MTC file with ddt4all or PyRen?

The question is that easy, But I don't think anyone will answer that

Marianpol, 2021-02-03:
PyRen generates mtcdata.txt file that contains car configuration when a documentation create function is used. To create a documentation you need to have DocDB_XX and BVMEXTRACTION folder that can be found in the CLIP files and then execute a following command.

doc_maker.py --vin VF1XXXXXXXXXXXXXX -L XX

where XX is your language e.g. GB

org07, 2021-02-03:
> vin VF1XXXXXXXXXXXXXX -L XX

vin VF1XXXXXXXXXXXXXX -L XX. Are we going to write the chassis number here?

Marianpol, 2021-02-03:
> vin VF1XXXXXXXXXXXXXX -L XX. Are we going to write the chassis number here?

You have to put your VIN number after the --vin argument (you write it instead of VF1XXXXXXXXXXXXXX).

org07, 2021-02-03:
> DocDB_YY ve BVMEXTRACTION

Where is DocDB_YY and BVMEXTRACTION this file?

org07, 2021-02-03:
Mr. Marianpol my goal is to make radio phase 2 it from the MTC file Phase 4 and write it back to the radio

Marianpol, 2021-02-04:
> Where is DocDB_YY and BVMEXTRACTION this file?
You can extract it from a CLIP DVD using extract.py from the PyRen or you can find it in the CLIP installation directory.
BVMEXTRACTION - C:\CLIP_X91\Data\GecAppli\Vehicles
DocDB - C:\CLIP_X91\Data\Documentation\Documentation.rar - unpack to a folder DobDB_YY

mtcdata.txt contains only vehicle configuration, e.g.:
`1234YF
2ROUDI
A5
ABLAVI
ABS
ACCAR2
AERA02
AFURGE
AIRBA2
AIRBDE
ALEAHL
ALOUC5
APL03
ATAR02
ATAV02
AVESM
AVFAP
AVGSI
AVOSP1
AVREPL
BANAL
BVA6
CA03
CDVOC1
CEAVRH
CHC02
CMAR3P
CPE
DANGMO
DG
DIESEL
DLIGM1
DPRPN
DPSEC
DRA
DRAP03
E06R
EA3
ECAMH1
ECOMOD
EQPEUR
ESPHSA
FDIU2
FEUAR2
FLTDIE
FPASS
FSTPO
HARM01
HAYMAN
HRGM01
HYB01
ISOFIX
ITPK6
KITGRV
KITPGR
KITPOU
KM
LAC
LAVPH
LEDH2
LFD
LVARAP
LVAVIP
MAPSUP
MLEXP1
PBNCH
PBRAC
PE2015
PHAS1
PNSTRD
POL01
PREALA
PREM01
PRIAC3
PRODIS
PROJAB
PROTBM
R9M
RAD65A
RALU18
RDIF10
RET04
RETRCR
RSGALT
RV
SAN408
SAN613
SAN913
SAN972
SANACF
SANBAT
SANCML
SASURV
SCCHBA
SCDCOF
SERIE
SEXTIN
SFIRBA
SGACHA
SGAR04
SGAV02
SNSTAP
SOP03C
SPFAIR
SPRDEN
SPRGAZ
SPRHMD
SRANCF
SREACT
SSAEBS
SSAPEL
SSAVSE
SSCABA
SSCALL
SSCCPC
SSDECA
SSDWGE
SSEQSH
SSFLEX
SSPTLP
SSRCAR
SSRIDO
SSTYAD
STANDA
STHPLG
SUSNPI
TATDB2
TCU0G2
TEGNE
TEMP
THABT1
TLRP00
TN
TR4X2
TRSEV1
VEC036
VLCUIR
VOLCHA
VT
XFD
XFD
XFD1TR`

You would like to edit a MTC file and do auto configuration of a radio from the Phase 4 which will be mounted in the Phase 2 car like CLIP does?

AlMeu, 2022-10-26:
@Marianpol 
I used the command as you described and get this:
[image]
What is wrong?

AlMeu, 2022-10-26:
Also this happen for a car 2022:
[image]

KarelSvo, 2022-10-27:
[VF1RFB00568913508.zip](https://github.com/cedricp/ddt4all/files/9875428/VF1RFB00568913508.zip)

AlMeu, 2022-10-27:
@KarelSvo 
Thanks I renew the database and find the VIN, but still ref.dat is absent and can't complete generation of html file.
How I can fix this?
How can I use DVD Reprog in pyren?

AlMeu, 2022-10-27:
@KarelSvo @Marianpol @shrlnm 
can you write me on my email numaialmeu@yahoo.com 
I need some help.

KarelSvo, 2022-10-27:
[VF1RFB00568913508.htm.zip](https://github.com/cedricp/ddt4all/files/9882518/VF1RFB00568913508.htm.zip)

AlMeu, 2022-10-28:
@KarelSvo 
Very thank you, for html file.
I want to make working also in my pyren for generate the file well without errors. Can you help me, please?
I have feeling that I am near, but still missing something.

shrlnm, 2022-10-30:
@AlMeu The latest CLIP version with complete BVM db is a v202. On the following versions they excluded REF.dat
DOC generation may work with any new BVM db, it not require REF.dat but only MTC.dat
DocDB_GB should be extracted next to the other DBs

AlMeu, 2022-10-30:
@shrlnm 
here is my email numaialmeu@yahoo.com for speak more.
I understood the tool is not finished yet. I will take BVM from v202 and try.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif