cedricp/ddt4all #532: need help please megane 2 reset uch

berendlevi, 2021-02-08:
hello i have a problem i have reset  uch no can i not start de car and not see the card key
i need help please

org07, 2021-02-08:
İstanbul'daysanız yapan birisini tanıyorum. Aynı zamanda adam anahtarlı aracı kartlı araca çevirmiş

https://www.youtube.com/watch?v=G9ZLhg35eGA 

https://www.youtube.com/watch?v=lkCz7QqYvbU&t=48s

https://youtu.be/k0UYZDSCRoA

cedricp, 2021-02-08:
You unfortunately reset your UCH to virgin mode... (there's a warning widget about that) The only way to recover is to find ISK (hope you requested it) and reprogram with cards. After that you'll need to set vehicle parameters.

berendlevi, 2021-02-08:
where can i find isk en where can i reprogram

berendlevi, 2021-02-08:
i have the log file

cedricp, 2021-02-08:
send the log file over

cedricp, 2021-02-08:
Megane 1 phase 1 or 2 ?

berendlevi, 2021-02-08:
megane 2 phase 1

berendlevi, 2021-02-08:
how can i sent the log file

cedricp, 2021-02-08:
put it on a server or https://pastebin.com/

berendlevi, 2021-02-08:
do you have email

berendlevi, 2021-02-08:
11:50:20.340;743;2180;61 80 82 00 36 40 23 04 41 4E 43 01 26 37 43 51 00 08 09 05 42 47 2F DB 3B 00 84 
11:50:20.720;7E0;2180;61 80 82 00 44 20 42 44 32 31 33 82 00 38 71 38 00 A7 74 00 11 0C 02 01 00 00 00 
11:50:20.971;74D;2180;61 80 82 00 30 60 33 44 30 30 31 00 00 30 0A 00 02 02 2A 00 00 00 41 65 02 00 00 
11:50:21.181;752;2180;61 80 82 00 41 10 04 08 32 31 39 39 35 00 01 02 00 19 12 00 08 01 01 30 12 00 00 
11:50:21.451;745;2180;61 80 82 00 35 11 82 4D 30 30 31 00 00 15 03 01 03 22 05 40 00 00 2D 15 14 00 40 
11:50:22.410;742;2180;61 80 82 00 24 66 31 08 44 5A 39 04 23 25 06 26 83 41 00 14 00 01 00 00 00 00 00 
11:50:26.981;742;2180;61 80 82 00 24 66 31 08 44 5A 39 04 23 25 06 26 83 41 00 14 00 01 00 00 00 00 00 
11:51:11.723;745;21 06;61 06 03 03 02 40 00 00 00 00 00 03 02 
11:51:20.040;745;3B 92;7B 92 
11:53:05.583;745;21 06;61 06 03 03 02 00 00 00 00 00 00 81 03 
12:05:36.904;745;21 06;61 06 03 03 02 00 00 00 00 00 00 81 03 
12:05:49.174;745;21AB;61 AB 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
12:05:50.881;745;21 08;61 08 00 00 00 00 00 00 0A 00 00 00 00 00 00 00 4B AD 17 38 00 00 00 00 00 02 00 00 02 92 00 00 02 8F 00 00 92 00 00 02 8F 
12:05:52.766;745;3B0501;7F3B22;Unknown
12:05:52.766;745;3B 05 01;ONG RESPONSE : Unknown 
12:05:56.281;745;3B0901;7F3B22;Unknown
12:05:56.281;745;3B 09 01;ONG RESPONSE : Unknown 
12:06:13.783;745;3B 08 10 20 20 10 81 88;7B 08 
12:06:53.732;7E6;3E00; 
12:06:55.235;7E6;3E00; 
12:06:56.734;7E6;3E00; 
12:06:58.239;7E6;3E00; 
12:06:59.736;7E6;3E00; 
12:07:01.238;7E6;3E00; 
12:07:02.741;7E6;3E00; 
12:07:04.234;7E6;3E00; 
12:07:05.743;7E6;3E00; 
12:07:07.242;7E6;3E00; 
12:07:08.737;7E6;3E00; 
12:07:10.228;7E6;3E00; 
12:07:11.567;7E6;14FFFFFF; 
12:07:11.809;7E6;3E00; 
12:07:13.239;7E6;3E00; 
12:07:14.750;7E6;3E00; 
12:07:16.251;7E6;3E00; 
12:07:17.754;7E6;3E00; 
12:07:19.248;7E6;3E00; 
12:07:20.755;7E6;3E00; 
12:07:22.246;7E6;3E00; 
12:07:23.751;7E6;3E00; 
12:07:24.468;7E6;14FFFFFF; 
12:07:25.254;7E6;3E00; 
12:07:26.754;7E6;3E00; 
12:07:28.253;7E6;3E00; 
12:07:29.754;7E6;3E00; 
12:07:31.248;7E6;3E00; 
12:07:32.748;745;3E00; 
12:07:34.250;745;3E00; 
12:07:35.747;745;3E00; 
12:07:37.259;745;3E00; 
12:07:39.011;745;3E00;7F3E12;Unknown
12:07:39.011;745;3E00;ONG RESPONSE : Unknown 
12:07:40.289;745;3E00;7F3E12;Unknown
12:07:40.289;745;3E00;ONG RESPONSE : Unknown 
12:07:41.621;745;3E00;7F3E12;Unknown
12:07:41.621;745;3E00;ONG RESPONSE : Unknown 
12:07:43.120;745;3E00;7F3E12;Unknown
12:07:43.120;745;3E00;ONG RESPONSE : Unknown 
12:07:44.783;7E6;3E00; 
12:07:46.282;7E6;3E00; 
12:07:47.782;7E6;3E00; 
12:07:49.283;7E6;3E00; 
12:07:50.805;7E6;3E00; 
12:07:52.294;7E6;3E00; 
12:07:53.806;7E6;3E00; 
12:07:55.294;7E6;3E00; 
12:07:56.804;7E6;3E00; 
12:07:58.298;7E6;3E00; 
12:07:59.805;7E6;3E00; 
12:08:01.300;7E6;3E00; 
[... 48 lines cut ...]
12:20:37.780;743;2180;61 80 82 00 36 40 23 04 41 4E 43 01 26 37 43 51 00 08 09 05 42 47 2F DB 3B 00 84 
12:20:38.336;74D;2180;61 80 82 00 30 60 33 44 30 30 31 00 00 30 0A 00 02 02 2A 00 00 00 41 65 02 00 00 
12:22:46.063;745;21AB;61 AB 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
12:22:47.657;745;21 06;61 06 03 03 02 00 08 00 08 00 00 81 03 
12:22:47.833;745;21 08;61 08 00 00 00 00 00 00 0A 00 00 00 00 00 00 00 4B AD 17 38 00 00 00 00 00 02 00 00 05 FC 00 00 05 F9 00 00 FC 00 00 05 F9 
12:22:56.904;745;3B0901;7F3B22;Unknown
12:22:56.904;745;3B 09 01;ONG RESPONSE : Unknown 
12:23:06.170;745;3B0501;7F3B22;Unknown
12:23:06.170;745;3B 05 01;ONG RESPONSE : Unknown 
12:23:22.373;745;3B 08 10 20 28 10 81 88; 
12:23:46.754;745;3B 09 00; 
12:31:03.048;745;3B 08 1A 71 88 12 01 82; 
12:32:39.996;745;21AB;61 AB 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
12:32:41.553;745;21 06;61 06 03 03 02 00 08 00 08 00 00 81 03 
12:32:41.723;745;21 08;61 08 00 00 00 00 00 00 0A 00 00 00 00 00 00 00 4B AD 17 38 00 00 00 00 00 02 00 00 07 91 00 00 07 8C 00 00 91 00 00 07 8C 
12:32:51.407;745;3B0501;7F3B22;Unknown
12:32:51.407;745;3B 05 01;ONG RESPONSE : Unknown 
12:32:59.051;745;3B 08 1A 71 80 12 01 82;7B 08 
12:33:03.027;745;3B0901;7F3B22;Unknown
12:33:03.027;745;3B 09 01;ONG RESPONSE : Unknown 
12:35:15.967;745;2180;61 80 82 00 35 11 82 4D 30 30 31 00 00 15 03 01 03 22 05 40 00 00 2D 15 14 00 40 
12:35:16.201;743;2180;61 80 82 00 36 40 23 04 41 4E 43 01 26 37 43 51 00 08 09 05 42 47 2F DB 3B 00 84 
12:35:16.746;74D;2180;61 80 82 00 30 60 33 44 30 30 31 00 00 30 0A 00 02 02 2A 00 00 00 41 65 02 00 00 
12:36:15.861;745;3B0900;7F3B22;Unknown
12:36:15.861;745;3B 09 00;ONG RESPONSE : Unknown

cedricp, 2021-02-08:
12:22:46.063;745;21AB;61 AB 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
-> ISK is already virginized here, sorry, can't help...
You'll need CLIP tool to reprogram it.

berendlevi, 2021-02-08:
oke now most the car to renault or not

cedricp, 2021-02-08:
depends on your knowledge, but I'd say yes

berendlevi, 2021-02-08:
i have also a log with elm.dtt

cedricp, 2021-02-08:
roughly the same data.

cedricp, 2021-02-08:
check if you have a line not filled with 0 for **21AB** frame

cedricp, 2021-02-08:
Didn't you see this wigdet ?

[image]

berendlevi, 2021-02-08:
yes i have check uch

berendlevi, 2021-02-08:
and virgine uch

cedricp, 2021-02-08:
I mean the warning about it will erase the UCH device ? I thought it was clear that using this plugin will cause data loss.

berendlevi, 2021-02-08:
you say remapping de uch

Marianpol, 2021-02-08:
Now if you want to use yours UCH and be able to add the keys back again you need to pull out the PIN from the ECM EEPROM (your ECU is Sagem/Valeo S3000). Another option is you can buy used ECM and UCH with keys from an another car and put it in you car. When you go for the second option you only need to configure the UCH module to match the original vehicle configuration.

berendlevi, 2021-02-08:
oke and when i if the pin what most i do next

berendlevi, 2021-02-08:
ddt4aLL give byy key a pin

cedricp, 2021-02-08:
@Marianpol Do you know where PIN is stored in the S3000 ? Have got one sleeping on a drawer :)

berendlevi, 2021-02-09:
Found ECU [Cluster/TDB] (not perfect match) :Tdb_BCEKL84_serie_4emeRev DIAGVERSION [4] SUPPLIER [ANC] SOFT [0008] VERSION [0905 instead 0615]

Marianpol, 2021-02-09:
> @Marianpol Do you know where PIN is stored in the S3000 ? Have got one sleeping on a drawer :)
> Think it's not necessary to pull out the EEPROM as there's a "read memory by address" request for it.

Unfortunately, I don't know the exact address where the PIN is stored. I've tried to compare a original and virgin EEPROM dump but I could not find any clear sing of the PIN. Some say that there is an ISK not the PIN.

read memory by address service is most likely to read a RAM (SRAM) memory of the Hitachi MCU (SH7055) not the EEPROM.
In a DDT database this service has static FF731A address. When we take a look at the memory map of this MCU we can see, that this address is probably pointing to an On-chip RAM address range. 

[image]

I've used this in service with dev session on my EDC16C3 with MPC555 MCU and addresses that I was able to read were exactly the same as the SRAM address range written in the memory map provided by a datasheet for this MCU.

So unfortunately, you have to open your ECU to pull the PIN/ISK out.

BarektE, 2021-02-09:
Actually, what you do is very simple. You can't do this with ddt4all, you will go to a Renault authorized service or a local mechanic with a Renault token will connect to renaulta with your chassis number online with the token and update the UCH, that's all, we do this a lot.

If there's a chance you'll come to me, I'll do it for free.

berendlevi, 2021-02-12:
come to you i live in the netherlands

berendlevi, 2021-02-13:
you can vin nr to pincode

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836 :|

closed 2023-06-02 by Furtif

Patriarch78, 2024-08-21:
> 12:22:46.063;745;21AB;61 AB 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 -> ISK is already virginized here, sorry, can't help... You'll need CLIP tool to reprogram it.

Can you help me? What ISK in my frame?
[image]

Patriarch78, 2024-08-22:
> This does not seem to be a Megane 2 UCH... (should be CAN protocol, not KWP2000)

This is Laguna II

biernatdamian9, 2025-03-21:
Pomożesz mi z Megane 2 i uch? Reset był zrobiony?