cedricp/ddt4all #534: Anyone know what MAX_TURN_POWER_LEVEL_X_ON_TM_CF (sec) does?

AndyIreland, 2021-02-11:
in the course of changing something I think I might have changed this accidently to '1' (1sec) or maybe not  in the BCM body control module settings and dont know what it should be set at by default (I know I should have took a picture of it beforehand , but unfortunately I didnt ) 

 MAX_TURN_POWER_LEVEL_X_ON_TM_CF (sec)
 
 the TM I should imagine = time  but that's all I know - does anyone know what it controls please?
 
 
[image]

AndyIreland, 2021-02-11:
UPDATE:
I looked in the Log and it looks like the parameter is '5' as its default .

would still like to know what MAX_TURN_POWER_LEVEL_X_ON_TM_CF (sec) does though if anyone knows

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif