cedricp/ddt4all #535: Writing acoustic parameters

uJay1980, 2021-02-12:
I'm referring to the closed issue #376 
After having swapped out my radio I would like to set up the acoustic parameters.
Unfortunately I'm not really clear about how I would be able to realize it.
I do have a switchable ELM USB interface, once purchased for use with forscan.
Of course pins have been resoldered in order to fit with Renault MS CAN.
By now I do get the common "serviceNotSupportedInActiveSession" error.
DDT4ALL is starting up with baud rate 500000. (Another interface always with 38400)
How to change it? Dropdown selection does not make any difference.
What about the session timeout value and the mentioned flow control?

Is there any chance to write the acoustic parameters using the actual equipment?
Or do I have to buy myself an OBDLink interface?

AndyIreland, 2021-02-12:
if you got a bluetooth obd2 adaptor then that is bound to not work over 38400 baud rate speed, but thats ok, thats fast enough for using.

do you get the green 'connected' bottom left of the screen?

Try 'can timeout' setting , try setting it to 400ms or 600ms.

what Renault is it you are trying to change the acoustic parameters on (ie clio megane etc) and what radio is fitted (ie rlink , medianav etc? ..)

KarelSvo, 2021-02-12:
uJay1980
You first need the data from an identical radio.
The disadvantage of sending the data via the Ecu Screen is that the Screen also carries out read requests that cause the Elm Buffer to overflow.
This problem does not happen with OBDLINK SX. 

I use OBDLINK SX and pyrene with ddt function.
Save dump from identical radio.
Convert Dump to Macro file.
If necessary, adjust some configurations in the macro and enter the current VIN with CRC.
Supplement Macro with Ecu address and Elm Init. Add SDS and reset commands.
Send macro automatically via pyrene terminal.

uJay1980, 2021-02-12:
USB adaptor, not bluetooth. My car is actually a Megane 3. 
Came with TomTom Live & Bose speakers, so that's the origin radio the parameters should be taken of.
Alternatively I think I should have also the parameters of a radio from a Megane 3 with R-Link and Bose.

I did a complete R-Link retrofit last year and a final mod lately adding a DAB radio from Renault Master.
(front panel modified to fit Megane radio slot and white button lighting)
All functions working as intended, configured properly to Bose/ext. amp.
Of course, the radio is configured with the Master OEM config by now.
Does not even sound bad, though.
But would like to finalize with setting up the right configuration strings

I do not have any experience with Pyren.
But I guess I should take a closer look.

KarelSvo, 2021-02-12:
Here you can find examples of Rlink 2 macros  https://gitlab.com/py_ren/pyren/-/tree/master/pyren/macro/rlink2

uJay1980, 2021-02-12:
@AndyIreland: Of course the green button connected was shown up correctly.

I gave it a try with a couple of different values for can timeout.
Unfortunately with still no success.
Like I said, I had been able to modify other settings, as well as for R-Link screen.

I guess it rather won't be possible to write these hex strings with my interface...

AlMeu, 2021-07-14:
The long parameters are big troubles in these units.
I can write these parameters on my desk.

hdano1, 2021-12-28:
> The long parameters are big troubles in these units. I can write these parameters on my desk.

Can someone ,who had success with writing accoustic parametwrs, please describe what cable, settings, version of windows ?

I accidentaly set factory reset on my R2_08 v3_81

I tried Win8, Win 7,  ELS27, Obdlink SX, Ford ELM327, ELM327  BT,  used DDt4ALL write option , pyren macro, but none of them can recover accoustic parameters.
Always errors.

By the way, there is possibility to set speed using some .exe program  for  ELS27 or OBDLIN SX to 115200, 500000 or 1000000. Which speed should be activated for the cable?

Ar4w, 2021-12-30:
@hdano1  big parameters are not the problem for usb connected devices. I use ordinary chinese usb elm327 for writing parameters with length up to maximum 4096 bytes. Usually I use serial line speed 230400 bod. 230400 is enough for almost all operations. Both programs ddt4all and pyren support "soft" configuring this line rate. In pyren you may set line rate with option key -r 230400. No need to use any additional programs for changing speed. Operation system does not matter at all
Regarding an errors you are getting, first check elm*.txt logs. Then check the length of parameter you are writing. The simplest way is to read it first and compare the length with what are you going to write. Some versions of ECU firmware may have the same parameter with a different length.
Also you may try to use --cfc option in pyren for change writing mode. It may helps.

Adel88888, 2022-07-21:
Hello i have also problem after some coding i got black screen and this i have repaired but now i cant hear and they cant hear me when i speak with Bluetooth and my voice control doesent work i have 
megan iv
VF1RFB00258202024
Com port 7
Elm usb 
Sistem is 3.3.16.946

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif