cedricp/ddt4all #537: Change Calibration in BCM module (its read only)

AndyIreland, 2021-02-15:
in an experiment say if I wanted to change REMOTE_LIGHTING_TM_TP (sec)  from 20 seconds to 30 seconds timeout :
[image]
and there is no 'write' available - how can I change this figure?

If i find the BCM95_SW25_V1.1_20150429T131442.xml folder and open it up in notepad.exe on my laptop in the DDT4ALL folder and find:
REMOTE_LIGHTING_TM_TP (sec) this is what I get:
[image]
is there anything i can change it that area to make it stay on 3osends instead of 20seconds? - if so what do I need to change or add? 

thanks.

AndyIreland, 2021-02-15:
> **in an experiment say if I wanted to change REMOTE_LIGHTING_TM_TP (sec) from 20 seconds to 30 seconds timeout :**
> [image]
> **and there is no 'write' available - how can I change this figure?**
> 
> **If i find the BCM95_SW25_V1.1_20150429T131442.xml folder and open it up in notepad.exe on my laptop in the DDT4ALL folder and find: REMOTE_LIGHTING_TM_TP (sec) this is what I get:**
> [image]
> **is there anything i can change in that area to make it stay on 30 seconds instead of 20seconds? - if so what do I need to change or add?**
> 
> thanks.

@KarelSvo @cedricp

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif