cedricp/ddt4all #538: Change ECU from Ethanol to Gasoline Clio MK3 1.2  2010

Fradenmk3, 2021-02-16:
Hi.
My Clio mk3 (2010) 1.2 flex fuel (75hp) is stuck on ethanol (alcohol) mode.
Is there any way to change this to gasoline  with ddt4all or PyREN?
There is gas in the car right now, but the little clio thinks its alcohol in it so the l/km is terrible (1.2l/10km)
I have looked around of hours with both of them, but i cant find anywhere to change this.
 

Or must this be done with Clip?

KarelSvo, 2021-02-16:
Show the result of scanning for your ECUs.

Fradenmk3, 2021-02-16:
[image]

KarelSvo, 2021-02-16:
There is no fixed choice between gasoline and ethanol in this ecu.
There is only this automatic learning procedure  https://www.docdroid.net/iQdyQus/sim32-pdf#page=242

Fradenmk3, 2021-02-17:
Thanks for the answer. I have tried too do this automatic learning procedure, but it still says that there is alcohol in the car.
But i found a lambda sensor that had no voltage at all, so i will try to change this next week when it arrives to see if i can to this learning procedure again.
At first the sensor did not listed in fault codes, so i didn't know where to begin.
I will tell if it works or not when i have change this sensor.

Thanks again for your help!

Fradenmk3, 2021-02-19:
Now i have changed the sensor. Now its says that i have 0.00% alcohol in the car, but it still say that is on the Alcohol mode.
But the consumption has been much better (0.5l/10km)

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif