cedricp/ddt4all #540: Hello I installed ddt4all latest version but it gives the error in the picture

org07, 2021-02-25:
Hello I installed ddt4all latest version but it gives the error in the picture
Also, it cannot find any ECU.

[image]

org07, 2021-02-25:
[image]

org07, 2021-02-25:
looking for a valiant to solve this problem

AndyIreland, 2021-02-26:
hello - try download from here it includes ecu and will put a shortcut on your windows desktop for DDT4ALL - if you want to use on Renault choose "All Cars" download hope this helps (should be safe but use site at your own risk): 

https://www.drive2.com/b/498093336985338243/

org07, 2021-02-26:
> merhaba - buradan indirmeyi deneyin, ecu içerir ve DDT4ALL için Windows masaüstünüze bir kısayol koyacaktır - Renault'da kullanmak istiyorsanız "Tüm Arabalar" indirmesini seçin, bunun yardımcı olacağını umuyoruz (güvenli olmalı, ancak siteyi kendi sorumluluğunuzdadır):
> 
> https://www.drive2.com/b/498093336985338243/

Thank you dear AndyIreland, already ddt4all and pyrene after ddt4all and pyreni are properly packaged, we are still downloading from Zmiterm blog. Nobody packs the update, they just patch

KarelSvo, 2021-02-27:
Your error message says, that the ddt2000 ECU database is missing. 
Cedric is not the legal owner of this database, so he cannot add it.
You must to find the database yourself and copy it into the ddt4all folder.
I am not using this ddt4all Python 3 update. Some functions are still faulty there.

org07, 2021-02-27:
> Hata mesajınız ddt2000 ECU veritabanının eksik olduğunu söylüyor.
> Cedric bu veritabanının yasal sahibi olmadığı için ekleyemez.
> Veritabanını kendiniz bulmanız ve ddt4all klasörüne kopyalamanız gerekir.
> Bu yeni ddt4all Python 3 güncellemesini kullanmıyorum. Orada hala bazı işlevler hatalı.

Can't download the latest version from here? I downloaded from here, it gives the errors I uploaded the picture above, it does not find any ECU

[image]

KarelSvo, 2021-02-27:
This download is without the Ecu database. I have already explained to you why this is so.

org07, 2021-02-27:
> Bu indirme Ecu veritabanı olmadan yapılır. Bunun neden böyle olduğunu size zaten açıkladım.
> 
> Ve bu güncelleme henüz tam olarak geliştirilmemiştir. Örneğin Wifi Elm çalışmıyor veya bayt 2180'e göre ECU tanımlaması yapmıyor.

Thank you very much KarelSvo for the information. If you weren't here, there is no one who has the skills to answer

AndyIreland, 2021-03-17:
am I allowed to mention this here - if not sorry and feel free to delete . the DDT4all with ECU's and a shortcut on your desktop can be downloaded from here (visit site and download at your own risk) 

https://www.drive2.com/b/498093336985338243/

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif