cedricp/ddt4all #541: rlink "Please add EOL settings"

AndyIreland, 2021-02-27:
it had to happen one day .. and that day is now it looks very much like - getting fed up with the rlink store greyed out as well as some other things greyed out even though i have a valid renault connected subscription  I threw caution to the wind and pressed configuration reset I think it were and pressed home 5 times to reboot and get this bloody death message! 

Gonna leave car disconnected all night in the hope that tomorrow it might reset itself but i dont hold out much hope - I reckon i have bricked the rlink.

 I have filled in the VIN details and as many parameters as I can remember because most of them were empty or disable by defualt now after the reset. 
 
 have i got any hope of getting it working again with DDT4all (or even PyREN I have that installed too) ? - or is it a visit to renault to load the firmware / software again on the radio to get it working again? - anything I can do at home any usb stick with firmware on that I could load back into the radio? 
 
 What does EOL mean - does it stand for 'End of Line' ? 
 
 Could I buy CANclip and get it working myself ? - my nearest reanault dealers is 50km away.
 
 Any Ideas most welcome please from anyone or even if @KarelSvo if your online and reading this - i know there are threads on here with people who had same problem and got same message on screen  but i havent read them all yet 
 
 many thanks. 
 Andy.
 
 
[image]

KarelSvo, 2021-02-27:
What if you
rlink eol settings
you write in the Google search engine?

mnigten, 2021-02-27:
Indeed end of line. Can be set with clip but most common way is to reload the firmware. For both the clip is needed. Sorry to say but the renault dealer is will be. 
I did a investigation even on russian sites but everywhere ended with the advice to go to the dealer.

AndyIreland, 2021-02-27:
> Indeed end of line. Can be set with clip but most common way is to reload the firmware. For both the clip is needed. Sorry to say but the renault dealer is will be.
> I did a investigation even on russian sites but everywhere ended with the advice to go to the dealer.

ah right thank you, thats a shame but if it has to be it has to be I suppose - thank you for answering

AndyIreland, 2021-02-27:
Have just been out to car just now , put in parameters , then hard reset and now have a beeping out of each door speaker in rotation now ... I feel a bit of progression no?

AndyIreland, 2021-03-01:
thank you all I managed to fix the rlink today . I used a TomTom 000 file on the SD card and filled in the parameters manually - done a hard reset and then radio started, asked for a security code , i inputted that and now radio works thankfully

LisovR, 2021-04-13:
> thank you all I managed to fix the rlink today . I used a TomTom 000 file on the SD card and filled in the parameters manually - done a hard reset and then radio started, asked for a security code , i inputted that and now radio works thankfully
If not a great secret, help other people solve this problem.

LisovR, 2021-04-13:
If not a great secret, help other people solve this problem.

AndyIreland, 2021-04-13:
OK , i dont know how much more i can explain - I used a tomtom.001 file on the SD card put it into the SD card reader of the car , filled out all the parameters with DDT4ALL including VIN number and CRC number and done a hard reset and then put in security code from my renault app and it worked for me

LisovR, 2021-04-15:
> OK , i dont know how much more i can explain - I used a tomtom.001 file on the SD card put it into the SD card reader of the car , filled out all the parameters with DDT4ALL including VIN number and [...]
Is this update file from the renault-connect website or something else?

LisovR, 2021-04-15:
Is this update file from the renault-connect website or something else?

AndreiBBF, 2021-08-31:
> OK , i dont know how much more i can explain - I used a tomtom.001 file on the SD card put it into the SD card reader of the car , filled out all the parameters with DDT4ALL including VIN number and [...]

AndryIreland, where did you find that file?

AndyIreland, 2021-08-31:
>> AndryIreland, where did you find that file? And what parameters did you insert and where? Thanks!

such a shame I cannot totally remember, but went into my bookmarks and it looks like I downloaded it from here - but cannot think this 100% 

https://disk.yandex.com/d/0VNd3V4L3UZEZX

and its got TomTom.000 after extracting it , so maybe thats it instead of TomTom.001 

Format the SD card to fat32 - put the extracted TomTom.000 file on it and no other file 

Fill out the VIN and CRC with DDT4all - try and fill out as many parameters as you can with DDT4all in the rlink section with the adapted cable / dongle and put the aftersales date to today's date  then hard reset the radio ... hopefully if it works the EOL warning should have gone and you would just have to put your radio code and away you go (hopefully) 

Sure if anyone still having problems I can see if i can remotely get it working again you will need: 

Windows 10 Laptop in the car  with good internet connection
a Bluetooth ELM327 which works with DDT4ALL (important it a BT one, or usb wired one and not a wi-fi one)
a special cable(or the ELM327 specially adapted) so you can read the rlink through the OBD2 socket.
This TomTom.000 file downloaded from that site and extracted onto a fat32 formatted 8gb-16gb SD card
and you with your car through the process so wen ready to see whats going on and reset the radio by pressing the home button. 
No guarantees it will work but its worth a try.

StepanenkoNadejda, 2022-10-16:
Так

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

elusivestudio, 2024-06-13:
@AndyIreland can you please share tomtom.000 file again on google drive or something so link doesn't expire. And also steps to perform to make it work again. I will highly appreciate it. Thank you!