cedricp/ddt4all #542: trying to read TCU - Bad ELM (7F3E21) messages

AndyIreland, 2021-03-01:
Hello can someone help me please - I have tried 3 different ELM's 2 Bluetooth ones and a wi-fi one. 

My Rlink Store is greyed out on my rlink as is my TomTom live services. - I have subscriptions to both. Both in date. 

I decided then to see what provisions services were activated on my TCU. 

DDT4all would not find the TCU when doing search with magnifying glass, so rather than blindly picking a TCU I loaded PyREN on my laptop and it found the TCU in my car of  TCU_GEN2_V3.2 DCM

So I chose TCU_GEN2_V3.2 DCM in DDT4ALL.  with my Bluetooth ELM327 with a PIC18F25K80 and go to Services to see if Internet Widget Store and TomTom Live services is all 'activated' but when I try to read all I get is red boxes and 'No Data' in the window it says 'Bad ELM (7F3E21) ' see pic:

[image]

[image]

[image]

[image]

and look they are all red:
[image]

so there is no way of telling what is activated or deactivated in the services - I am almost sure this maybe why the icons are greyed out on my R-link and cannot be accessed? 

Here is what PyREN reports all of them deactivated!! :( 
[image]

here is log files - i cannot understand them or what they mean , maybe they throw some light on the matter?

[screens.txt](https://github.com/cedricp/ddt4all/files/6058497/screens.txt)

Many thanks for help I really want to get these activated if I can

KarelSvo, 2021-03-01:
> here is log files

No ddt4all and pyrene log files available, only screens.txt.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif