cedricp/ddt4all #543: Which UPC-EMM select in DDT4ALL on Renault Trafic 2020

SukiSan84, 2021-03-01:
Hi

Car: Renault Trafic 2.0, year 2020

I'm trying to get an access to lighting of the car. 

Over PC (win 10) I run DDT4ALL. Program can normally connect over OBD2 Wi-Fi interface. Then select car, Trafic III, and when I go on section UPC-EMM, I got a list of...hmmm...I think modules or different models of EMM (see the attachment).
[image]

And now, when I select one of this from the list, DDT4ALL says "DISCONECTED". But files can be read further on. Does anyone have any clue:
1. which UPC-EMM should I select (Is this written somewhere in car, like physically? Or there are some software to recognise version?) and 
2. why I’m getting feedback, that is disconnected?

SukiSan84, 2021-03-02:
LOGs from DDT4ALL
[Just double click on selected EMM.zip](https://github.com/cedricp/ddt4all/files/6068366/Just.double.click.on.selected.EMM.zip)
[selecting two EMM and then lighting.zip](https://github.com/cedricp/ddt4all/files/6068367/selecting.two.EMM.and.then.lighting.zip)

From Car Scanner got info:
OBD-II
ECU name: ECM-EngineControl
Calibration ID: 237617069S
Calibration Verification Numbers (CVN): Mi

Engine control unit
Diag. ver.: 86

When Selecting EDISON and Lighting I got this picture:
[image]

When Selecting T4_EMM and Lighting I got this picture:
[image]

KarelSvo, 2021-03-03:
> LOGs from DDT4ALL

No Elmtest result

AndyIreland, 2021-05-17:
are you not searching for the ECU's using the magnifying glass when connected?

SukiSan84, 2021-07-17:
Sorry for long reply. Yes, I ‘am searching with magnifying glass. The procedure is like this:
1. in FileExplorer: C:\Python27\ddt4all I start ddt4all.py
2. select Wi-Fi connection
3. put OBD2 adapter in car (Renault Trafic)
4. In DDT4ALL click on "Connected mode"
5. Then I try different options:
  a) Select the car from the list and then click SCAN and select an option CAN. CMD result what was checked is in attachment. Check the file "Selecting the car and click SCAN and then CAN-100proc.txt"
  b) Select the car from the list and then click SCAN. CMD result what was checked is in attachment. Check the file "Selecting the car and click SCAN.txt"
7. The problem is that I still have quite a big list of UPC-EMM.....and I don't know which one I should select. Even if I select randomly one or another, I got feedback on the bottom of DDT4ALL program, red light "Disconnected".
[image]
[image]
[Selecting the car and click SCAN.txt](https://github.com/cedricp/ddt4all/files/6834615/Selecting.the.car.and.click.SCAN.txt)
[Selecting the car and click SCAN and then CAN-100proc.txt](https://github.com/cedricp/ddt4all/files/6834616/Selecting.the.car.and.click.SCAN.and.then.CAN-100proc.txt)

AndyIreland, 2021-07-17:
thats not how you should be doing it - you should start it up and with green 'connected' at the bottom left you should not select your vehicle in the left hand box but go straight to magnifying glass and then select 'CAN' in the middle window and then after 100% of waiting it should find all your ECU's for your car. There are only a few times on some ecus if they are not found by magnifying glass that you can put your vehicle into the box and manually select a random ECU but that way is the most dangerous

KarelSvo, 2021-07-18:
> 4\. In DDT4ALL click on "Connected mode"

Why not click on "ELM benchmark" when you have Elm problems? Then give me the elm_ddt file from the logs folder.

KarelSvo, 2021-07-18:
> you should start it up and with green 'connected' at the bottom left you should not select your vehicle in the left hand box but go straight to magnifying glass

No, first select the vehicle type and then scan.
But first the Elm has to be checked, when the connection to the Elm breaks.
The green status only shows the status of the connection: software to Elm, not Elm to Ecu.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif