cedricp/ddt4all #544: Can't connect over BT ELM327 - invalid literal for int() with base 10 error

szumial, 2021-03-02:
Hi! I am having an issue with a Bluetooth ELM327 adapter + modified cable. The BT module is an iCar VGATE2 which I have seen some colleagues working with. The DDT4All screen flashes and disappears after selecting bluetooth connection. I managed to take a screenshot of the error message. Interestingly, I can connect to the program when selecting the standard USB cable icon, but then the ECUs are empty and the program does not read any values. What could be the issue?

I am using the latest DDT4All version from the releases section and my car is a Laguna 3 PH3.
[image]

KarelSvo, 2021-03-03:
> I am using the latest DDT4All version from the releases

This version still has various bugs.

szumial, 2021-03-03:
Can you suggest which release should I use instead?

KarelSvo, 2021-03-04:
https://www.drive2.ru/b/498093336985338243/

tomasz-soltysik, 2022-02-27:
This issue was fixed in #511. There is no release yet with this change, but you can clone repo or download it.

meerkovo37, 2022-02-27:
> This issue was fixed in #511. There is no release yet with this change, but you can clone repo or download it.

How would I go about downloading this to solve this issue, just had this happen with my Bluetooth ELM scanner today

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif