cedricp/ddt4all #545: need EV features settings for Renault zoe rlink MFD

AndyIreland, 2021-03-03:
hello - i recently pressed reset configuration on my rlink on my Zoe EV and it erased all settings. I have since restored all my settings and the rlink is working and everything but there is one setting I cannot remember the ID# for the EV features 

anyone know what they are. Type2was 'not configured' I remember that but  'vehicle receptable type' was ID# something and 'Type 1 cable' was ID# something. 

[image]

I did back up my settings as  zoe22.ecu but how do you open back up a .ecu file in DDT4ALL? 
Thanks.

KarelSvo, 2021-03-03:
Just use Pyren in ddt mode for your experiments. You can create a Bachup file beforehand. You can use this file to compare or restore configurations.
Unfortunately, ddt4all does not offer this option.

AndyIreland, 2021-03-10:
> Just use Pyren in ddt mode for your experiments. You can create a Bachup file beforehand. You can use this file to compare or restore configurations.
> Unfortunately, ddt4all does not offer this option.

Hello @KarelSvo  - I managed to fix my MFD Radio and thats working OK. 

However i am very interested in this backup and restore method in PyREN and want to restore my settings for my TCU unit in my Zoe EV. 

So , I have found the backup from last November in the Dumps folder in pyREN folder of my laptop 

1605566518_TCU_GEN2_V3.2_20151109T114222.txt 

As you can see it has a .txt extension - how do I restore that back into my TCU by PyREN or DDT4ALL and would I have to change the extension to .XML ? 

Thanks

KarelSvo, 2021-03-11:
Pyren automatically converts the read commands into write commands.
You should use a current version of Pyren for this.  https://gitlab.com/py_ren/pyren

AndyIreland, 2021-03-11:
> Pyren automatically converts the read commands into write commands.
> You should use a current version of Pyren for this. https://gitlab.com/py_ren/pyren

Thank you for your reply @KarelSvo - is there a step by step guide anywhere for backing up the .txt (or would it have to be changed to .xml?) files anywhere , because i am stumped how to? . 

Do I rename the 1605566518_TCU_GEN2_V3.2_20151109T114222.xml to 1605566518_TCU_GEN2_V3.2_20151109T114222.txtold ??
.. and then rename the 1605566518_TCU_GEN2_V3.2_20151109T114222.txt to 1605566518_TCU_GEN2_V3.2_20151109T114222.xml and then choose 'prefer to read from xml file' in pyREN? is that how I reload the backup?

AndyIreland, 2021-03-11:
Ok , i have found the section in PyREN for 'rollback' so I put the November last year TCU .txt file in the Dump folder and selected it then in 'expert mode' and then loaded it 

- is that it, should it have loaded the backup from last year now to the TCU?

KarelSvo, 2021-03-11:
What Pyren sends and receives is written in 2 log files. Read commands start predominantly with 22 ....... and write commands with 2E ........

MacManiac78, 2021-08-04:
Hi there! Im glad i found this topic about Pyren. I managed to get a non working TCU after an Renault update. The flashdrive was bad... and than i got the message the update failed. So i tried again and it failed again. Since than i noticed soms apps where greyed out and later i found out the TCU did stoo working. Now Renault wants to change the TCU for €840 euro. I got the MFD working again by removing the fuse. But the TCU never did came back online after removing the fuse trick. I just think some setting are corrupted. Would it be possible to upload a setup file of another ZOE to get my TCU working again?

AndyIreland, 2021-08-04:
> Hi there! Im glad i found this topic about Pyren. I managed to get a non working TCU after an Renault update. The flashdrive was bad... and than i got the message the update failed. So i tried again [...]

only problem is that it would have different VIN and CRC setting in it ... now , whether you can just change it and add your own VIN and CRC into it and away it would work is something I wouldnt be 100% of . 
I think there was 2 different fuses on the Zoe on the interior glove box fuses. one for MFD/Radio/Rlink and then another one for the TCU unit if i am correct - what year and model Zoe you have @MacManiac78  ?

MacManiac78, 2021-08-04:
Dear Andy,

I have pulled both fuses allready. This did not reset it. I also went to the dealership. A TCU update was done. They contacted Renault Netherlands. 3 things might be possible. I had some isseus with Smart Charging. Before Renault launched the Smart Charging app. (Now Renault Mobilize) a was Smart Charging with the Jedlix app on my ZOE INTENS ZE40 Q90 05-2017. All worked fine until Renault rebranded the app and the app was "updated". Than Jedlix told me my ZOE was blacklisted by Renault. So i was mad.....   Maybe Renault did cut off the TCU or... the TCU died or....  the TCU has been corrupted due the failed update.... but i really doubt that. Traffic is working. Apps do not work....   the charge planner does not work either. So I think....   Renault blocked the TCU or....   some setting has ben  the TCU has been corrupted due the failed update. Before i contact Renault i would like to try and upload new settings. I have tried to change some settings allready, but the TCU does not write it in its memory. I have noticed that when i close DDT4ALL and reopen it...  all settings are gone. I also noticed....   when i close DDT4ALL and than pull the fuses....   some settings are still in place. I think....   Renault changed some other settings....   somewhere higher....   CANbus based maybe. If you could help me out i would be so gratefull. - Ben

AndyIreland, 2021-08-04:
> Dear Andy,
>
> I have pulled both fuses allready. This did not reset it. I also went to the dealership. A TCU update was done. They contacted Renault Netherlands. 3 things might be possible. I had some [...]

Hello Ben - if you can get internet in your car on a win10 laptop and hook up to DDT4ALL with a bluetooth or USB OBD2 adaptor its possible i might be able to . I could hook up by remote desktop and take over your laptop and then have a look at the TCU and MFD settings myself and see if there is something that stands out to me?

AndyIreland, 2021-08-04:
> > 
> 
> You don't want to give your ecu_ddt and elm_ddt files for research?

who's that aimed at @KarelSvo ? - me as in as I shouldn't be giving out my ddt files?

MacManiac78, 2021-08-04:
i have no problem with sharing a log of some kind. I did not have time to install pyren yet. Might do that this werkend.

AndyIreland, 2021-08-04:
> i have no problem with sharing a log of some kind. I did not have time to install pyren yet. Might do that this werkend.

ah right well if you need a hand and I am around you can give me a shout and I can see if I can help you   if you like - i just have a feeling they are greyed out because settings are not activated in the TCU and mfd ecu's

MacManiac78, 2021-08-04:
Andy, that is my feeling too. I think by updating the settings where not changed, only a newer version of the software with the same settings.

AndyIreland, 2021-08-04:
also take note that normally as well if you have activated all the settings in the TCU for internet apps and charge schedular and all that lot that you normally have to press the home button on the r-link radio 5 times to give it a reboot and then see if they are still greyed out

MacManiac78, 2021-08-08:
I think i found it. (I tried PyRen....  but this does not do anything for me, it is less evolved than DDT4ALL is...  at least...  for what i have seen....  I also found DDT2000, but that does not for me YET)  So I tried DDT4ALL again....   I think...   the TCU has been reset and it it being set to GEN2. I think my Zoe Q90 Intens 2017 has a GEN1 TCU. I have also noticed ALL fields in the TCU section have "NO DATA" So i think all settings are deleted.

KarelSvo, 2021-08-10:
lots of text, but no ddt4all log files

MacManiac78, 2021-08-10:
Because i do not how to create a log. I'm sorry.

KarelSvo, 2021-08-10:
ddt4all writes the logs automatically
the files are in the program folder

AndyIreland, 2021-08-10:
in the case of 'Dumps' PYREN will only make a dump of the radio if the OBD2 adaptor is wired up for, or a splitter lead is wired up for, so that it will read the multimedia through the CAN socket . If you just have plain old OBD2 in the OBD2 socket (not adapted OBD2) then it will not back up a 'Dump file' of your r-link - may even go for logs as well I am not sure

MacManiac78, 2021-08-11:
Karel, thanks i will look into that. If i have older logs, would DDT4ALL be able to use that old log for reprogramming too? I will post them asap.
Andy, i have split the OBD2 wiring into a working Y-cable.

MacManiac78, 2021-08-11:
Found it!
[screens - old.txt](https://github.com/cedricp/ddt4all/files/6967210/screens.-.old.txt)
[elm_ddt - old.txt](https://github.com/cedricp/ddt4all/files/6967213/elm_ddt.-.old.txt)
[ecu_ddt - old.txt](https://github.com/cedricp/ddt4all/files/6967214/ecu_ddt.-.old.txt)
Hope this helps

shrlnm, 2021-08-13:
@MacManiac78 
Below is a  pyren/mod_ddt dump generated especially for your VIN.
[000000_MFD_v5.2_20181029T155922.txt](https://github.com/cedricp/ddt4all/files/6982901/000000_MFD_v5.2_20181029T155922.txt)

First look at diffs "dumps/Show diff" in mod_ddt
Then you may try to roll it back. "dumps/rollback"

MacManiac78, 2021-08-13:
@shrlnm 🥇 THANK YOU! 🥇 
I have downloaded the files.  I'm trying to update PyRen because it does not show the needed path.

MarcosMar, 2021-08-21:
Hello guys!
I've follow @AndyIreland  and all dtt4all entusiasts for a long time. 
I have a Zoe Q90 22KW.
I made some mods with dd4all and all working great, no issues.
6 months ago i put the watch and temp on Table du Bord (Cluster). all great until i change in TDB to "sans NAV" (No MFD). 
No i dont have precoditioning, remote conditioning, remote charge schedul......
I already put back this setting but the problem but..... same problem.
Is there anyone with the same problem?
Have reached a solution?

Sorry my bad...... bad english

MacManiac78, 2021-08-21:
Dear Andy, I can not get PyRen to work. Would you help me with Remote Desktop?

KarelSvo, 2021-08-21:
> Dear Andy, I can not get PyRen to work. Would you help me with Remote Desktop?

which language and for which operating system?

MacManiac78, 2021-08-21:
Dutch Win10

KarelSvo, 2021-08-21:
[to test for you](https://transfiles.ru/6s6rj)

[virus test](https://www.virustotal.com/gui/file/e7045dbad5619ac380c89c12266346761718ad4e64b5a816fd8c23407b6a34af/detection)

AndyIreland, 2021-08-24:
> Dear Andy, I can not get PyRen to work. Would you help me with Remote Desktop?

Hello @MacManiac78 - sorry for late reply only seen your message, , yep if you want i can take a look remotely and have a look at the settings and see if it looks all as it should

MacManiac78, 2021-08-24:
Dear Karel, i have downloaded this Pyren version. Thanks. (I have not tried it on the Zoe yet.) It is completly different then the one i had downloaded before.
Dear Andy, if this does not work, i will contact you.

AndyIreland, 2021-08-24:
> Dear Karel, i have downloaded this Pyren version. Thanks. (I have not tried it on the Zoe yet.) It is completly different then the one i had downloaded before.
> Dear Andy, if this does not work, i will contact you.

no problem @MacManiac78  - if you do need me too, just give me a shout

KarelSvo, 2021-08-24:
> It is completly different then the one i had downloaded before.

It's a self-installing version, with all the necessary modules for your Zoe.
If you still want to download the detailed diagnostic instructions, specially extracted for your chassis number. The link is only valid until tomorrow morning.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif