cedricp/ddt4all #546: Safely resetting the TCU with DDT4ALL

AndyIreland, 2021-03-07:
Hi, I want to have a go at resetting the X91 TCU in my Renault Zoe and i know I could hard reset it by pulling the fuse in the interior fusebox and that I can choose 'Hard Reset' in the session management/session - but i am wondering what the other 2 resets do 'reset configuration parameters'  and 'reset test parameters' ?  because i once pressed a reset on the rlink radio in DDT4all and it semi bricked the radio with a yellow triangle saying 'fill in EOL settings' so that has made me wary now 

[image]

ElTurco19, 2021-04-05:
Hello Andy 

I don't know why you want to reset but be wary. DDT4All is too powerful and you can easily make an irreparable mistake as our friend Cedric repeats time and time again. 

I advise you to disassemble the 2x battery terminal, remove the battery from its location, wait 5 minutes and touch the +/- terminals of the vehicles together for a few minutes (be careful without the battery or without touching the battery). 

-be that it will help and prevent you from making a mistake. 

good luck.

AndyIreland, 2021-04-06:
> Hello Andy
>
> I don't know why you want to reset but be wary. DDT4All is too powerful and you can easily make an irreparable mistake as our friend Cedric repeats time and time again.
>
> I advise you to [...]

Thank you for answering. i managed to get the TCU working OK again by using rollback .txt backup in PyREN and it appears to be working now ... albeit 10pm every night LOL I have a connection and off in the daytime most of the day  , but its most probably Renault servers playing up or something

SedatAklar, 2023-05-28:
Hello. It blocked my Talisman TCU from satellite even though I have a reno subscription. I can't get live data. It gives a no network connection error. How do I reset it safely? I do not have a backup txt that was previously taken. Should I follow the battery removal method? Enjoy your work.

AndyIreland, 2023-05-28:
> Hello. It blocked my Talisman TCU from satellite even though I have a reno subscription. I can't get live data. It gives a no network connection error. How do I reset it safely? I do not have a backup txt that was previously taken. Should I follow the battery removal method? Enjoy your work.

Hi, There are a few different TCU's showing up for the Talisman  in DDT4ALL What is your one showing up in DCM Renault in DDT4ALL?

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

SedatAklar, 2023-07-08:
Hello, sorry for the delay. 2016 1.6 dci 130 hp, Rlink2 v2.2.19.300 micom 4992 for small screen 7 inch DDT4ALL on DCM Renault when I search for ecu's with a magnifying glass it finds TCU_ GEN2_V3.2. I have activated many deactivated settings related to TCU and saved them. Actives are registered in DDT4ALL. But nothing changed in Rlink2. Still no network. Is there a way to get the TCU working? Thanks for the answer.
[image]

SedatAklar, 2023-07-09:
Merhaba, gecikme için özür dilerim. 2016 1.6 dci 130 bg, Rlink2 v2.2.19.300 micom 4992 kçük ekran 7 inç için DDT4ALL'da DCM Renault'da genişlemele ecu'ları arattıklarıda TCU_ GEN2_V3.2 yi buluyor. TCU ile ilgili birçok devre dışı ayarını aktif yaptım kaydettim. DDT4ALL da aktifler kaydedilir. Ancak Rlink2 de hiçbir şey olmadı. Hala şebeke yok. TCU' yu çalışır duruma getirmenin bir yolu var mıydı? Cevap için Teşekkürler.
[image]

AndyIreland, 2023-07-09:
if you have no network then maybe your rlink subscription needs renewing maybe?

SedatAklar, 2023-07-09:
Hello, thanks for your interest. I have RLink subscriptions. Renault customer services told me that "Your data connection and TCU has been cancelled." But as we can see in the attachment my subscriptions goes as well for now. I think the issue is due the TCU. How can i activate TCU again with DDT4ALL?
[image]

SedatAklar, 2023-07-09:
Hello, thanks for your interest. I have RLink subscriptions. Renault customer service told me "Your data connection and TCU have been cancelled". However, as you can see in the appendix, my subscriptions continue for now. I think the problem is with TCU. How can I re-enable the TCU with DDT4ALL?
[image]

matzel82, 2023-07-09:
Please connect to your dealer. He can check your active service. I work at a Renault Garage.... sometimes possible the services make some troubles

AndyIreland, 2023-07-09:
>  But nothing changed in Rlink2. Still no network.

You might have to check your communications settings on your rlink too. 

you have an rlink2 so I dont know about that one, but have a look at this. My Zoe EV has an rlink1 so in the 'navigation' ECU I have the settings of APN USER PASSWORD DNS1 DNS2 MCC & MNC (obviously all filled in on my car) so if you have them all filled in and corectly set up you might then get the network?

[image]

SedatAklar, 2023-07-13:
Hello, I will try this. But I don't know what the correct information to write is. The dealer doesn't do it for legal reasons anyway.

AndyIreland, 2023-07-13:
this is settings for mine, but as I say mine is a rlink1 evolution - your settings might differ if its a rlink2 
[image]