cedricp/ddt4all #551: clearing dtcs leads to crash under Linux

picalke, 2021-03-22:
Hey

I encountered an error once I tried to clear DTCs from varoious ECUs (espace IV ph1 parking brake and megane II ph2 engine ecu).

The error seems to be caused by some hex/int/float arithmetics. Traceback:
`
Traceback (most recent call last):
  File "/home/piotr/ddt4allPy3/parameters.py", line 1428, in clearDTC
    options.elm.set_can_timeout(options.cantimeout)
  File "/home/piotr/ddt4allPy3/elm.py", line 804, in set_can_timeout
    val = hex(val)[2:].upper()
TypeError: 'float' object cannot be interpreted as an integer
Aborted (core dumped)
`
My Python is Python 3.8.8
Kernel Linux zbylut 4.4.0-206-generic #238-Ubuntu SMP Tue Mar 16 07:52:37 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
OS is a bit ancient Ubuntu 16.04

chrizbee, 2021-10-06:
I had the same error just now. You can resolve it by changing line 801 in `elm.py` to
```python
val = round(value / 4)
```
You can also replace `round()` with `int()` if you don't care about accuracy.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif