cedricp/ddt4all #553: Win 10 cant connect to ELM through bluetooth

axney, 2021-03-28:
Other applications can connect to ELM372 but ddt4all cant.

Here is the benchmark:

Trying to open portCOM3 @ 115200
Serial<id=0x2c0841d0a60, open=True>(port='COM3', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=5, xonxoff=False, rtscts=False, dsrdtr=False)
Trying to open portCOM3 @ 38400
Error:could not open port 'COM3': PermissionError(13, 'Access is denied.', None, 5)
ELM not connected or wrong COM port COM3
ELM reset...
****************************************
*       Connection to ELM was lost

ElTurco19, 2021-04-05:
Hello Axney you should use an adapter with USB, BBFly or Vint brand is a good product. Good luck.

Koleosguy321, 2021-04-08:
Hey, I had a similar issue with the bluetooth ELM327 only very recently. Where it said "no data" was being shown.

Bluetooth ELM327 seems to have an issue with DDT4ALL, no idea why but it is just the case.

WIFI ELM327 seems to work fine and for me, USB ELM327 worked fine (after downloading the windows drivers).

majki09, 2022-05-05:
Hi @axney. I've had similiar issue with my BT ELM dongle. I debugged the python code and found out that my dongle's name is not ELM but OBDII. I modified the code and now I have everything working.
My mod is here: #679

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif