cedricp/ddt4all #557: DDT4All NO DATA

Koleosguy321, 2021-04-02:
Hello, so I'm trying to active Android Auto on my rlink2 system on my 2017 Koleos, but when I connect using Bluetooth ELM 327, it shows all "NO DATA" in the audio section. I have 09/07/2019 database

Anyone know why is this?

ElTurco19, 2021-04-05:
Hello Koleos you should use an adapter with USB, BBFly or Vint brand is a good product. Good luck.

Koleosguy321, 2021-04-06:
> Hello Koleos you should use an adapter with USB, BBFly or Vint brand is a good product. Good luck.

Thanks for the help. I ordered a USB elm327, so I'll try that and if it doesn't work then I'll try one of those brands you mentioned.

ElTurco19, 2021-04-06:
Hi,

be careful to choose a chip Pic18f24k80 or better Pic18f2480 and also the quartz 4000 mhz. The ideal would be a 2x faster stn1170 but no worries with the Elm 

good luck.

Koleosguy321, 2021-04-08:
> Hi,
> 
> be careful to choose a chip Pic18f24k80 or better Pic18f2480 and also the quartz 4000 mhz. The ideal would be a 2x faster stn1170 but no worries with the Elm
> 
> good luck.

Hey so the eln327 USB one was even worse, didn't even detect it on my ddt4all.

I need to get obd MX+ for a different car. You reckon that one will work fine?

Could you link me one that will work?

closed 2021-04-08 by Koleosguy321