cedricp/ddt4all #558: bunları nasıl aktif ederim megan 2

tcebe, 2021-04-02:
[image]

org07, 2021-04-03:
:) 
bunlar aktif zaten

org07, 2021-04-03:
sen buncacık bilgiyle arabayı bozarsın çok fazla karıştırma hatta hiç karıştırma benden söylemesi

tcebe, 2021-04-03:
org07 sende fazla bilgi varsa anlat bi zahmet bilgimiz olsun ?
aktif olsa çalışır herhalde pyren programından baktıgımızda bu ekran cıktı..

ElTurco19, 2021-04-05:
Hi tcebe you should already start by greeting people and writing in English so that everyone understands and supports you.

Selam tcebe, öncelikle bir yazıya başlarken insanları selamlamak en az terbiye gerektiren bir bilgidir. Sonra ingilizce sorarsan daha çok insan anlar ve yardım eder sana.

org07, 2021-04-05:
1 tanesinin örneğini vereyim o klima isteği devre dışı demesi senin klimanın kapalı olduğunu söylüyor klimayı açtın mı orada KLİMA aktif yazacak onun için diyorum samimi diyorum sen o programlarla arabanı kurcalama kesin bozarsın ve baya bir maliyetli olur.  Programın yazarı **Shrlnm** Türkçe bilseydi seni bu forumda banlardı ;)

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif