cedricp/ddt4all #559: Problem with the Talisman display !

ElTurco19, 2021-04-05:
Hello to all Git members. 

First of all a big Respect to Cedric for sharing his very useful DDT4ALL software

to Zmiterm for Pyren 

ans Shrlnm for Scripts and experience sharing. "You are unforgettable kings" 

I have a Talisman and following an update made by a friend, the screen has turned black. While doing some research i came across this fabulous script from friend Shrlnm whom i thank and wish him a long happy life. 

I tried the scripts one by one as described and the screen to work at the last script. Only i have a different model of Talisman, the screen shows a Scenic model and i am missing some features. 

How can i fix this problem ? 

Thanks in advance and really sorry for the long post. 

Long happy life to everyone.

KarelSvo, 2021-04-07:
An interface with STN 1170 chip has 2 CAN channels that can be selected via software commands.

ElTurco19, 2021-04-16:
Hi KarelSvo 

Thank you for the information, actually i saw a relay module assembly for Ford software, the person uses a stn1170 but he has the possibility to choose the mode (manual with switch) or (automatic with a relay) which is not available on DDT4ALL. 

If you have a tip i could try and share the diagram and the end result. 

Thank you.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif