cedricp/ddt4all #560: Clio 2 1.2L 2000 Ecu Sagem Safir 2 Don´t connect to ECM, DDT4ALL say: Reposta ELM: BLOCKED

wilfredocumare, 2021-04-07:
Hi,
I have a Clio 2 1.2 8v year: 2000 with ECU Sagem Safir 2, and i don find any form to connect to the ECM with nothing ELM327 App, just with DDT4ALL i found connect to AirBag, UCH, Climate, but don´t with ECM, Doing automatic search for ECUs only detects the ones I mentioned above, but when I manually select the Safir 2 ECU,these messages appear:

Pedido ELM bloqueado:3350
Resposta ELM : BLOCKED
ECU uses SDS 10C0
Pedido ELM bloqueado:3350
Resposta ELM : BLOCKED
ECU uses SDS 10C0
Pedido ELM bloqueado:3350
Resposta ELM : BLOCKED
No ReadDTC request for that ECU
ECU uses SDS 10C0
Pedido ELM bloqueado:3351
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3354
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3352
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3350
Resposta ELM : BLOCKED
ECU uses SDS 10C0
Pedido ELM bloqueado:3351
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3354
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3352
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3350
Resposta ELM : BLOCKED
ECU uses SDS 10C0
Pedido ELM bloqueado:3351
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3350
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3152
Resposta ELM : BLOCKED
ECU uses SDS 10C0
Pedido ELM bloqueado:3351
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3350
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3152
Resposta ELM : BLOCKED
ECU uses SDS 10C0
Pedido ELM bloqueado:3351
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3354
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3352
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3152
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3350
Resposta ELM : BLOCKED
ECU uses SDS 10C0
Pedido ELM bloqueado:3351
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3354
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3352
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3152
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3350
Resposta ELM : BLOCKED
ECU uses SDS 10C0
Pedido ELM bloqueado:3351
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3352
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3350
Resposta ELM : BLOCKED
ECU uses SDS 10C0
Pedido ELM bloqueado:3351
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3352
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3350
Resposta ELM : BLOCKED
ECU uses SDS 10C0
Pedido ELM bloqueado:3351
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3350
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3352
Resposta ELM : BLOCKED
ECU uses SDS 10C0
Pedido ELM bloqueado:3351
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3350
Resposta ELM : BLOCKED
Pedido ELM bloqueado:3352
Resposta ELM : BLOCKED

What this means?
Thanks in regards for help

KarelSvo, 2021-04-07:
For the ISO8 protocol on Safir you need ddt2000 and a KKL interface.

wilfredocumare, 2021-04-07:
> For the ISO8 protocol on Safir you need ddt2000 and a KKL interface.

Thanks for the answer.
KKL interface is the same VAGCOM? Because I have one of these blue 409.1KKL VAG cable, and i tried to connect with DDT2000 but it don't recognize that interface, I don't know if i did something wrong.
Thanks in advance.

ElTurco19, 2021-04-16:
Hi friend yes that's correct, as mentioned by KarelSvo, you have to open to see the chip if CH340 or Ftdi, if Ftdi just choose the right com port and Choose "Com Port" mode if you have a chip CH340 then you have to change the USB baudrate to 10400 baud and of course install the CH340 driver. 

Good work.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif