cedricp/ddt4all #563: update rlink-2 v3 to v9 and error many item

shahram241, 2021-04-17:
Hi, I updated version 3 to version 8 and then to version 9, but my R-link is not working properly. Can you help me?
https://uupload.ir/files/47px_1.jpg
[URL=https://www.uplooder.net/files/4ba44d8e48f239a55e394205879b9282/3.mp4.html]3.mp4[/URL]

shrlnm, 2021-04-18:
Hi
What is micom version? If you did not upgrade it in an official garage then I afraid I can not help you. You may only try to downgrade it back to original version 3

shahram241, 2021-04-18:
hi @shrlnm 
What is micom version?
I upgraded myself with unofficial software: version 3.3.6.980 to 9.0.34.616
Please help me?

shrlnm, 2021-04-18:
micom version should have 4 digits. 5276 for example

shahram241, 2021-04-18:
version is 5276

shrlnm, 2021-04-18:
Then there is no way. 5276 is incompatible with LGU v7 and upper.

shahram241, 2021-04-18:
Where should I go?

shahram241, 2021-04-24:
I was able to reduce the R-link version to 2.2.19.301
But after the upgrade, it turned black to version 3 pages and remained in boot mode.
You can guide me
https://uupload.ir/files/a2gv_2.2.jpg
@shrlnm

shrlnm, 2021-04-26:
@shahram241 did you downgrade to 2.2.19.301 or to 3.3.16.986?

shahram241, 2021-04-26:
@shrlnm Yes, I downgraded to  version 2, but when I tried to return to the original version, the screen turned black

shahram241, 2021-04-26:
@shrlnm I installed the files you gave me, but the screen is still black.The air conditione turns on automatically
cmdr_radnav_high_repair.py
mod_elm.py
mod_globals.py

shrlnm, 2021-04-26:
If these scripts repaired the box before but now not help, then I afraid the box is dead (

shahram241, 2021-04-26:
@shrlnm You gave these scripts to Virgin mode.
Can a new script help?

shahram241, 2021-04-26:
@shrlnm The old scripts had 290 errors when running

shahram241, 2021-04-27:
@tina2004 Hello, has your problem been solved?

shrlnm, 2021-04-27:
I can help only after virginizing or configuration reset. Unfortunately I can not help after wrong upgrade/downgrade procedure.

shahram241, 2021-04-27:
@shrlnm After downgrade to the lower version 2.2.19.300, I clicked on this option and the screen turned black
1.Clear DTC
2.Configuration Reset
3.Hard Reset
https://uupload.ir/files/v6uk_2.2.jpg

shrlnm, 2021-04-27:
@shahram241 if it was reset then give me your email and I try to help

shahram241, 2021-04-27:
@shrlnm shl241@yahoo.com

shahram241, 2021-05-19:
Hello
I can install this unit for my car, is it compatible?
Talisman 2017
Thank you
RADYO MULTİMEDYA BEYNİ LANR 16 ORİJİNAL LG 281156817R 281154385R 281157255R

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
RADYO MULTİMEDYA BEYNİ LANR 16 ORİJİNAL LG 281156817R 281154385R 281157255R

Renault Megane 4, Talisman, Kadjar Radyo Beyni Uygun Fiyata Bonpar'la Hemen Kapınıza Gelsin.
 |

 |

 |

closed 2021-06-12 by shahram241