cedricp/ddt4all #565: Cruise control/Speed limiter disabled.

Snade90, 2021-04-22:
Hello,

I recently had a problem with my ABS on my 2014 latitude, and it was giving me "check brakes" alert on my dashboard, and also disabled my cruise control/speed limiter functionality.

Long story short after replacing the ABS module the alert is gone and I've successfully cleared all DTC errors and they are not coming back, but the cruise control and speed limiter functionality remains disabled.

I have tried to check the dashboard ECM for my latitude on ddt4all and the option shows that cruise control and speed limiter is enabled. I have tried to disable and re-enable it using ddt4all but to no avail as the functionality remains disabled inside the car when I try to press the cruise control or speed limiter switch (a light is supposed to turn on in the dashboard but nothing happens)

Does anyone have any ideas on how I may solve this problem using ddt4all or otherwise?

Please help and much appreciated!

Mikescotland, 2021-04-22:
Some of the Ecus might disable the limiter / cruise control when the switch is not in a rest position when doing anything on the ecu.
Try set the switch in rest (all off position) and then. 
Switch ignition without starting the engine 
1. Activate it on BCM 
2. Reactivate it on ecu.

Then switch off ignition and try if the cruise / limiter works.
Any error codes left? Some of them would prevent cruise / limiter to be active.

Snade90, 2021-04-22:
@Mikescotland 

Thank you very much! 

I was not aware of pyclip!

I downloaded the app on my phone, and selected the ECU reset function inside the Injector ECU while ignition was on and engine was off.

Everything works again!

Can't thank you enough, saved me a lot of trouble!

Mikescotland, 2021-04-22:
Glad I could help. Happy driving!

ohayyam, 2021-04-22:
@Mikescotland  " Here's what manual says about this issue (using can clip of course) "   Which manual and can you share this manual?

Mikescotland, 2021-04-22:
> @Mikescotland " Here's what manual says about this issue (using can clip of course) " Which manual and can you share this manual?

From Renault Dialogys database. They are different for each model and ECU.
I'm not sure if I can attach here as they are copyrighted.
But if you give me your model and what ecu you want to troubleshoot, I'm happy to send in pdf

ohayyam, 2021-04-22:
Talisman, injection : I will be glad if you can send an e-mail omern7100@gmail.com

NoUsername10, 2022-07-18:
> From Renault Dialogys database. They are different for each model and ECU. I'm not sure if I can attach here as they are copyrighted. But if you give me your model and what ecu you want to [...]
.

@Mikescotland 
Hi Mike.
I will try here as it seems you have knowledge of Renault and the documentation to boot.

Hopefully you have time :-)

I have Renault Clio III 2012 and I have retrofitted CC.

I have Renolink 2.0 (from actual developer) and ddt4all.
I have the settings required and all buttons work (tested and verified with Renolink)
(+,-,C,R, CC, CL) and I have attached a extra switch on the clutch on the green connector.

The thing is, I can’t get the settings to save. 
All commands are marked with Success.
Im now all out of ideas.

Setting is found under TBD(Tbd_X85ph2_V2)

Do you have any ideas what I could be doing wrong?

Mikescotland, 2022-07-18:
Do you have any, and I mean, any fault code with this clio?
Clio has a nasty thing, for some fault codes it disables the limited /
cruise control.
I did not try to activate this function in ddt4all, but in renault can clip
(obviously a clone).
In TBD function LC061 must be configured to with.
Check if injection ECU sees the cruise control / limiter Under DCM 3.4
module and parameters LC165-168

Stupid question. But did you mount the switch near the hand brake to
actually activate cruise control / limiter? Without it, steering wheel
buttons don't do anything and if nothing is connected to the switch near
the handbrake, clio throws a fault code.

NoUsername10, 2022-07-18:
> Do you have any, and I mean, any fault code with this clio? Clio has a nasty thing, for some fault codes it disables the limited / cruise control. I did not try to activate this function in ddt4all, [...]

Hi @Mikescotland 
Thank you for your reply!

I don’t have any fault codes, the switch under the handbrake is mounted and the buttons work. 

[image]

Everything here works and I get the correct feedback on all the buttons.

I don’t have clip, is it possible to write the codes you suggested with ex Renolink or ddt4all? 

Here are options from Renolink:
[image]

Mikescotland, 2022-07-18:
Can you take a screenshot of DCM34 > DIV > Cruise control?

Mikescotland, 2022-07-18:
Ok sorry I've just had the screenshot now. It seems it's not active in the
injection ECU.
Give me a minute, trying to find out where it sits. I'm afraid I don't know
French...

Mikescotland, 2022-07-18:
Ok. I just reviewed it all in ddt4all. It looks like for engine ECU ddt4all
doesn't have configuration mode. I remember it was the case when I replaced
injectors, to code them I needed to buy a clone of Canclip. The screen you
attached is for cluster meter but the function remains off on the ECU
computer. Now, mine is a diesel one so my ECU is DCM34, yours might be
different and it might have options to activate CC /limiter. Can you give
me the ECU type on ddt4all for your petrol engine? I assume essence is a
petrol engine!

NoUsername10, 2022-07-18:
> Ok. I just reviewed it all in ddt4all. It looks like for engine ECU ddt4all doesn't have configuration mode. I remember it was the case when I replaced injectors, to code them I needed to buy a [...]

Ah, ok that explains a lot. 
Thank you for taking the time to look at this!

I also have a diesel, k9k so this is probably the issue then (and me being in the wrong place in the software) 😀

Here is my scan with all the “ecu’s”

[image]

Mikescotland, 2022-07-18:
Ok. That explains a lot. Diesel has the newer ECU DCM34, you have a
face-lift 2012 model (so is mine). This injector ECU is supported very
poorly in the ddt4all (and derivatives like PyREN). To be able to change
anything I had to buy a CanClip.
Try to borrow from someone who has it. I'd help but I'm in Glasgow.

NoUsername10, 2022-07-19:
> Ok. That explains a lot. Diesel has the newer ECU DCM34, you have a face-lift 2012 model (so is mine). This injector ECU is supported very poorly in the ddt4all (and derivatives like PyREN). To be [...]

Thank you again for taking the time to help!
I live in Sweden, else id happily pay you to enable it.

Been sitting in the car for days with different OBD Wi-Fi, Bluetooth, and finally a USB sx, bough different softwares / apps and searched all over internet, but only you could provide the correct information and why it doesn’t work!

Lesson learnt for me,(even though I like to tinker with stuff) just buy the more expensive OBD that you know works.

Again, hope you have a nice day and know that I’m thankful!

NoUsername10, 2022-07-20:
@Mikescotland 
Hi again!

After some more tinkering and with the directional help from you I got it working without Can Clip!

I used Pyren and Clip database.
Finally found the instructions here: 
[Activate CC with Pyren](https://www.drive2.ru/l/468361717936554808/) 

We can enter “write mode” in Pyren with “cmd” command inside the ECU to get to the “Expert menu” where it’s possible to activate the CC.

So, we don’t need a Clip, Pyren can do it for you if you have the Clip database.

Sensation when the light came on was fantastic 😀

[image]

Mikescotland, 2022-07-20:
Hello
I'm glad you've managed to do it via Pyren. I tried with injectors so many
times and gave up and bought a CanClip (cost me €100). This was the reason
I moved from pyren to canclip as I could do everything without digging into
some expert modes etc.
If you can, post this method in the forum, I found some people trying to do
the same with Megane, failed with ddt4all.
I'm not familiar with pyren so much (I use mobile version called pyclip to
monitor / change some parameters on the go).
Sorry I couldn't be of more help.

Mikescotland, 2022-07-20:
BTW you must be rich! Half of the tank of diesel with todays prices! 😛

NoUsername10, 2022-07-21:
> BTW you must be rich! Half of the tank of diesel with todays prices! 😛
> […](#)

Diesel is now a luxury item so we are now in the upper class driving a diesel 😀 
(if you weren’t already) 
But yeah, it sure is a pain to fill it up.

You did absolutely the right thing in my opinion.
To anyone who want to do more stuff than basic clearing of codes / monitor values, just buy Clip.
It’s the way to go.

With all the other solutions it’s all a path with small breadcrumbs to find until you reach the solution (if you ever do). Very time consuming.

Also, no need to be sorry at all, your OBD codes and knowledge of the ECU helped me to move on from what I tried (DDT2000) to the next thing (Clip) so for me I wouldn’t have solved it without your help!

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif