cedricp/ddt4all #566: black screen

tina2004, 2021-04-26:
Hello all!
After an update, my R-link2 8.7" Talisman cannot boot and now I got a black screen. 
aircondetion on.
Please help!

shrlnm, 2021-04-26:
Hi. 
What is your micom version? What was a previous LGU version? How did you do an update?

tina2004, 2021-04-26:
hi shrlnm
micom version 5276.
previous version 2.2.19.300
rlink store.
when i upgraded my r-link with usb disk, the usb  had error and upgrading  is not compeleted now my car isnt work truly

shrlnm, 2021-04-26:
unfortunately 5276 is not compatible with LGU higher then V3.3 
For upgrading your r-link to v8 you had to upgrade MICOM first by help of official garage. 
I afraid your r-link can not be recovered now by well known methods  ((

tina2004, 2021-04-26:
@shrlnm There is no official garage in our area. You can help me? Plaese

shrlnm, 2021-04-26:
Sorry, I can't. Even if it was mine, I'll better bought a new one.

tina2004, 2021-04-26:
@shrlnm How can I run Emergency download?

shrlnm, 2021-04-26:
https://github.com/cedricp/ddt4all/issues/123?fbclid=IwAR13txzW6m5f-_yGau64cXGMiA9Ij6w6UCvyCyDBgkx9yboxXRCvgoIRXVE#issuecomment-564513977

tina2004, 2021-04-26:
@shrlnm How do want run Emergency download?
on/off  key + home key +
vol up + vol down

shrlnm, 2021-04-26:
I've tried this and it was enough to hold + and - buttons on joystick and long pressing Power button next to the screen

tina2004, 2021-04-27:
I can not enter the emergency download

shrlnm, 2021-04-27:
It is predictable ((. Emergency download not helps in case of real "black death".

AlMeu, 2021-04-28:
If it was blocked in wrong upgrade is very bad.
The solution remains to change the unit with one without VIN and antitheft code, and transfer if possible the settings from your unit to new one.
If you need give me an email and I can help in this.

tina2004, 2021-04-28:
@AlMeu cenatina2004@gmail.com.

mfarramadan, 2021-08-03:
Hello, Is there a way to downgrade

AlMeu, 2021-08-03:
Why you want do this?

mfarramadan, 2021-08-03:
I upgraded from 3.15 to 3.16 and now i have problems with my AA maps navigation so want to downgrade again.

AlMeu, 2021-08-03:
Android Auto maps doesn't have anything to do with firmware upgrade. It's related by firmware of mobile.

mfarramadan, 2021-08-03:
I read a lot of forums and this is a none bug in this version and not only for google maps but also with waze. Do you know how to downgrade?

AlMeu, 2021-08-03:
I don't know what you read and what you belive, but in all unit firmwares v3.3, 7.0, 8.0, 9.0 for R-Link 2 Android Auto is running the same. And I have tested all of these on my units. When a phone is running well, is running well in all firmwares. When doesn't want run because of phone problems will not run in anyone of these.

mfarramadan, 2021-08-03:
Thanks !
I managed to downgrade to 3.16.96 instead of 3.16.98 and it is working fine now.
I am using a s20+

ilker-50, 2021-11-13:
Merhaba 
Aynı sorun bendede Mevcut yardım ederseniz sevinirim teşekkürler
ilk_naz_@hotmail.com

TomaszMrozekPBL, 2022-03-19:
> It is predictable ((. Emergency download not helps in case of real "black death".

Hello, I read some topics and in most of them people wrote about You. Can You help me with this problem: I want to change radio for new one. I bought fabric new one from 2020 year of course without VIN number. After install new radio, and fill options in rad nav what should I do to have active screen because its black screen. I dont see nothing, but using obd I have connections witd radnav and can change there options. I have pyren, should I use this macro for screen size? https://gitlab.com/py_ren/pyren/-/tree/master/pyren/macro/rlink2 or there is another macro or way?

ilker-50, 2022-03-22:
Merhaba

Ben aynı sorunu kontrol unitesini değiştirerek çözdüm maalesef değişmeden olmuyor...

Android için Outlook<https://aka.ms/ghei36>'u edinin

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif