cedricp/ddt4all #567: R-Link 2 black scree

Goran-Barznje, 2021-04-26:
Once i did some reset and then the screen was not switch on, we can say the screen changed to black and nothing working 

Kindly i need your help for fix that

Note: i just use this picture by DD4all[image]

shrlnm, 2021-04-27:
Give me your email

shahram241, 2021-04-27:
@shrlnm After degrading to the lower version 2.2.19.300, I clicked on this option and the screen turned black
1.Clear DTC
2.Configuration Reset
3.Hard Reset

Goran-Barznje, 2021-04-27:
> Give me your email

Goran909090@gmail.com

Appreciate for your help

testeur70, 2021-05-16:
Hello all,
I just faced the same problem after entering in the developper mode in my 2018 Kadjar rlink2. I unfortunately pushed high resolution display and my rlink2 turned black and can not switch on. Radio is still working but black screen. My Rlink2 is not virginized I just pushed high resolution in this hidden menu. 
I read on forums that DDT or Pyren with its macros can solve this black screen. Can anyone confirm? Do I have to modify my obd connector to connect to my radio or a my normal interface is enough. 
Thanks for helping if you faced a similar problem.
Thanks

testeur70, 2021-05-18:
hello,

I solved my problem with DDT4all but havn't changed any parameters. I just had to validate  in the engineering page  the "central dispalu parameter" frame with the same parameters. Only push "Write display" in expert mode. Nothing has been changed. After this a reset (pushing for 15 seconds the radio on/off buton and the screen is now working. 

[image]

ilker-50, 2021-11-13:
Merhaba
Aynı sorun bendede var yardımcı olurmusunuz teşekkür ederim.
 ilk_naz_@hotmail.com

igno77, 2022-01-05:
@shrlnm  I need to know one thing. I've a macro made by you to wake up and configure a new R-Link unit and it works. There is a macro to wake up only the R-Link without put inside any configuration? May I use the macro for rotate the screen or activate AA and CP? I need only to switch on a new RLink without put inside any data. My mail is igno_v @ hotmail .com
Thanks a lot

Constantine328, 2022-01-21:
Hi there,

Unfortunately, I have exactly the same problem as Goran-Barznje...
@ shrlnm can you help me with my problem too?
my email is sebastian328@web.de

Many Thanks

gabrielcozma33, 2022-09-18:
> @shrlnm After downgrade to the lower version 2.2.19.300, I clicked on this option and the screen turned black
> 
> 1.Clear DTC
> 
> 2.Configuration Reset
> 
> 3.Hard Reset
> 
> https://uupload.ir/files/6uns_22.jpg

Did you solve it ? I have the same problem

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

615TK, 2025-03-16:
Hello everyone, good evening, I increased the screen scale in the Megane 4 R Link multimedia developer mode and the screen went black. Bluetooth and radio are actively working but the screen is off. What can I do about it? With which device can I load the screen parameters? I would be very happy if you could help me. Best regards.

shrlnm, 2025-03-17:
> Hello everyone, good evening, I increased the screen scale in the Megane 4 R Link multimedia developer mode and the screen went black. Bluetooth and radio are actively working but the screen is off. What can I do about it? With which device can I load the screen parameters? I would be very happy if you could help me. Best regards. [@shrlnm](https://github.com/shrlnm)

Hello @615TK You may install qpyren3 and implement builin macro. If it doesn't help then read https://www.drive2.ru/b/643019689764615344/

615TK, 2025-03-17:
@shrlnm  First of all, thank you very much for your response. I will look into the matter. Best regards.