cedricp/ddt4all #569: Config low beam not to turn on when high beam at daylight and gearbox upshift

marcuscl, 2021-05-06:
Hello, is it possible to config the headlight so when it's day the xenon light doesn't turn on when i use high beam? Only when xenon isn't normally turned.

And does anyone know if it it possible to program the gearbox's upshift at custom rpm when full throttle?

Hope it is the right place to post this..

AndyIreland, 2021-05-07:
hello - I think we all need more information than that before you get answers , like what is the model and Year of your Renault?

marcuscl, 2021-05-15:
Anyone know if its possible to program the headlights to do so? :))

AndyIreland, 2021-05-17:
thanks - when you search with the magnifying glass what BCM /UCH ECU does it find?

marcuscl, 2021-05-17:
Thanks :)
[image]

AndyIreland, 2021-05-17:
thanks, Ah right I see you are using the android ported ECU tweaker and not the PC DDT4ALL programme right then anyway for the lights there seems to be not much you can change in the exterior lights section but try 'dual bulbs' setting see if that does what you want - on later cars / ecus / software you have an awful lot more you can configure for the exterior lighting 

[image]

AndyIreland, 2021-05-17:
freeshift is in the Cluster TDB area - I dont think you can customise it , just have the upshift show or not on the dash - the TDB cluster is showin as red in your pic so be very careful with changing those settings do at your own risk:

[image]

marcuscl, 2021-05-18:
Thanks for your help! :) I will try it.

Another thing.. do you know if i can disable the auto ignition off when the car is stopped and I open the drivers door? :))

AndyIreland, 2021-05-20:
> Thanks for your help! :) I will try it.
> 
> Another thing.. do you know if i can disable the auto ignition off when the car is stopped and I open the drivers door? :))

no I am actually trying to figure that out in my Renault Zoe EV it turns the radio off each time I open up the drivers door with the engine off - infuriating , but it must be to save power and not run down battery!

marcuscl, 2021-05-20:
> no I am actually trying to figure that out in my Renault Zoe EV it turns the radio off each time I open up the drivers door with the engine off - infuriating , but it must be to save power and not [...]

I have the same "problem". I had the radio out and measured the wires, but didn't find anything... Maybe the drivers door sensor can be disabled or something..  

Let me know if you find out anything. I'm also trying figure it out :)

AndyIreland, 2021-05-20:
> I have the same "problem". I had the radio out and measured the wires, but didn't find anything... Maybe the drivers door sensor can be disabled or something..
>
> Let me know if you find out anything. [...]

will do - i am wondering now if i can short out the relay in the interior fuse box that clicks when drivers door opens to try and keep everything alive, i know that will mean that the electronic steering lock will not lock  and everything will stay on but thats what I want to try and achieve

marcuscl, 2021-05-20:
> will do - i am wondering now if i can short out the relay in the interior fuse box that clicks when drivers door opens to try and keep everything alive, i know that will mean that the electronic steering lock will not lock and everything will stay on but thats what I want to try and achieve

Thanks :)

I actually want the same. I want the radio, steering lock and everything else to lock and turn off when i lock the car, not when i open the door.

AndyIreland, 2021-05-20:
> Thanks :)
>
> I actually want the same. I want the radio, steering lock and everything else to lock and turn off when i lock the car, not when i open the door.

indeed! - thats the way it should be ... and in fact thats what does happen so I dunno why that the radio and windows have to turn off as soon as you open up the drivers door , because as you say everything goes off when you lock the car anyway.

marcuscl, 2021-06-04:
> indeed! - thats the way it should be ... and in fact thats what does happen so I dunno why that the radio and windows have to turn off as soon as you open up the drivers door , because as you say [...]

Hello, i have found out that the radio on/off is controlled by the can if you have the radio with navigation(tomtom in mine), so there must be a way to change it with obd... but i don't know how.

AndyIreland, 2021-06-05:
> > > >Hello, i have found out that the radio on/off is controlled by the can if you have the radio with navigation(tomtom in mine), so there must be a way to change it with obd... but i don't know how.

Thats what I am hoping too - but been into every setting in DDT4ALL that I can find for the MFD navigation and nothing there I can see - there is 'Door State' and i have set that to enable and disable but still no difference

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif