cedricp/ddt4all #570: Looking for standard settings of TCU EV Configurations if anyone can check please?

AndyIreland, 2021-05-06:
Hello - if anyone got a Renault Zoe EV (or a renault with a TCU fitted of TCU_GEN2_V3.2 ) I am looking for your settings showing up please? I didnt make a backup of mine or took a screenshot before changing some of the settings and i cannot remember what they should be. Thanks. 

[image]

shrlnm, 2021-05-07:
I may try to help but It is better by email

AndyIreland, 2021-05-07:
> I may try to help but It is better by email

thank you for that @shrlnm  - I have sent you my email address on drive2.com

Darlio, 2021-05-17:
Hello, I have a Zoe last generation... By email is indeed a good idea... C U soon Andy from Ireland ;-)

Colmrenault, 2022-02-03:
Hi Andy,
Just wondering if you every managed to get the settings. I bricked my 2017 Clio 4 TCU. Trying to get a second hand one to work. It turns out the second hand TCU is from a 2017 Zoe.
Colm Ireland

AndyIreland, 2022-02-04:
> Hi Andy, Just wondering if you every managed to get the settings. I bricked my 2017 Clio 4 TCU. Trying to get a second hand one to work. It turns out the second hand TCU is from a 2017 Zoe. Colm Ireland

hi Colm - I did get mine working again. my TCU in my Zoe is TCU0_Gen1_V3.4 - it could be different TCU in the Clio 4 , especially being 2017 because my Zoe is 2013.

Colmrenault, 2022-02-04:
Yes i managed to change the VIN and CRC.  I'll keep googling i guess.

AndyIreland, 2022-02-04:
> Yes i managed to change the VIN and CRC. I'll keep googling i guess.

Did you do a dump text file in PyRen by any chance of your old one so you can do a roll back ? 

if you have a bluetooth or wired usb ELM327 OBD2 adaptor want me to have a look in the settings of your TCU and can bring a windows 10 laptop out to the car and get internet on it I can maybe look at it remotely and see what is going on if you like. Just give me a shout if you want me to

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif