cedricp/ddt4all #571: I cant connection to ecu

Ba2aribakisli, 2021-05-11:
Hi,I connected my obd2 device but when ı tried connecting any ecu app has freeze on the initializing CAN protocol section.Can you help me ?

[image]

KarelSvo, 2021-05-13:
Do an Elm test with ELM327 Identifier https://play.google.com/store/apps/details?id=com.applagapp.elm327identifier&hl=tr&gl=US
Show the RAW data from this test.  If ATPPS is not supported, your Elm is probably not suitable. 
Also read this documentation on poorly copied Elm interfaces https://cvtz50.info/en/elm327/

Ba2aribakisli, 2021-05-13:
I did elm test and my device not supported ATPPS.
Thanks for help.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif