cedricp/ddt4all #572: Memory Error on creation of ZIP database/json files

fware-de, 2021-05-15:
Hello,
I got a ddt2000 v 02.2021 database and wanted to create the JSON/ZIP files from it.
I get memory errors on different files when I run it. See below.

I also do not see and ZIP file in the destination folder during process. So I do not know where the files are located.
System/OS shows now lack of memory for any process.

############### output from app ########################################

Starting processing ecus\C1A_Run2_BCM_DDT_76_v2_20200904T162338.xml 631/3487 to C1A_Run2_BCM_DDT_76_v2_20200904T162338.json
Traceback (most recent call last):
  File ".\ddt4all.py", line 603, in zipdb
    parameters.zipConvertXML(filename)
  File "C:\Program Files (x86)\ddt4all\parameters.py", line 1691, in zipConvertXML
    layoutjs = dumpXML(target)
  File "C:\Program Files (x86)\ddt4all\parameters.py", line 1514, in dumpXML
    return dumpDOC(xdoc)
  File "C:\Program Files (x86)\ddt4all\parameters.py", line 1650, in dumpDOC
    return json.dumps({'screens': js_screens, 'categories': js_categories}, indent=1)
  File "C:\Program Files (x86)\ddt4all\Python27\lib\json\__init__.py", line 251, in dumps
    sort_keys=sort_keys, **kw).encode(obj)
  File "C:\Program Files (x86)\ddt4all\Python27\lib\json\encoder.py", line 209, in encode
    chunks = list(chunks)
MemoryError

fware-de, 2021-05-29:
Additon.

on Python 3.x the transfer to JSON does not work at all.
Immediate error "cannot be serialized" or so.

So ZIP creation is dead.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif