cedricp/ddt4all #573: Renault Espace 2016 config for ACC (adaptative cruise control)

mmitech, 2021-05-15:
I was reading the other issues about activating ACC on cars that already have all the necessary parts (proximity radar and LDW) but they are really long and have so much unrelated discussion, so I thought I'd open a new issue to get the latest info about this.
 
I am wondering if anyone was able to activate the ACC and how they did it?

robertalks, 2021-05-16:
Im watching this, I have the same issue with my megane. I have the steering wheel and activated with ddt4all, but doesn't seem to be enough.

gonzaloalbito, 2023-03-21:
I'm about to buy an Espace V from 2016 aswell. It lacks ACC and LDW. It has emergency brake assist and park assist so I'm not sure it has all needed sensors (cameras and proximity sensor) to enable LDW and ACC after buying it, or it will not be possible at all. Could anyone help me?

closed 2023-06-02 by Furtif