cedricp/ddt4all #574: Cannot Clear my DTC codes with DDT4ALL

AndyIreland, 2021-05-17:
**i made some changes with DDT4ALL so decided to clear the DTC codes pressing on the heart icon but it does not seem to be clearing them - have tried with ignition off , and then also in park mode (selector in drive , keycard in slot , long press on start button) but it still didnt clear out of the window. Here is copy of the info in the DTC window:** 

-----------------------------------------------------------------------------------------------------

ECU trouble codes
DTC #0
DTCStatus.pendingDTC : [00] No
DTCStatus.testFailedSinceLastClear : [00] No
DTCDeviceAndFailureTypeOBD : 37849 [0x93d9]
DTCFailureType.ManufacturerOrSupplier : 15 [0xf]
DTCFailureType.Category : [08] Bus Signal / Message Failures
DTCFailureType : [8f] erratic
DTCStatus.testNotCompletedSinceLastClear : [01] Yes
DTCStatus.testNotCompletedThisMonitoringCycle : [00] No
DTCStatusAvailabilityMask : 57 [0x39]
DTCStatus.warningIndicatorRequested : [00] No
DTCStatus.confirmedDTC : [00] No
StatusOfDTC : 16 [0x10]
DTCStatus.testFailedThisMonitoringCycle : [00] No
DTCStatus.testFailed : [00] No
DTCDeviceIdentifier : [93d9] USB Host Presence not detected

DTC #1
DTCStatus.pendingDTC : [00] No
DTCStatus.testFailedSinceLastClear : [00] No
DTCDeviceAndFailureTypeOBD : 44545 [0xae01]
DTCFailureType.ManufacturerOrSupplier : 5 [0x5]
DTCFailureType.Category : [05] System Programming Failures
DTCFailureType : [55] not configured
DTCStatus.testNotCompletedSinceLastClear : [01] Yes
DTCStatus.testNotCompletedThisMonitoringCycle : [00] No
DTCStatusAvailabilityMask : 16 [0x10]
DTCStatus.warningIndicatorRequested : [00] No
DTCStatus.confirmedDTC : [00] No
StatusOfDTC : 16 [0x10]
DTCStatus.testFailedThisMonitoringCycle : [00] No
DTCStatus.testFailed : [00] No
DTCDeviceIdentifier : [ae01] PPP Data Communication Authentication

------------------------------------------------------------------------------------------------------

KarelSvo, 2021-05-17:
I use only  pyren for error diagnostic . pyren  shows errors that are relevant for the respective ecu version. Ghost errors are filtered out. ddt4all does not have this filter.

AndyIreland, 2021-05-17:
> I use only pyren for error diagnostic . pyren shows errors that are relevant for the respective ecu version. Ghost errors are filtered out. ddt4all does not have this filter.

thanks - i will try seeing what PyREN says - i just thought when you change stuff in DDT4ALL you should clear DTC codes with DDT4ALL

KarelSvo, 2021-05-17:
If you want to do it professionally, you go with pyren before and after the change and document the error and configurations with the help of the dump function. 
At the same time, you will receive a backup of your configurations, with which you can repair your ECU in an emergency. 
Here is a description of how you can create an error manual for VIN.  https://www.drive2.com/l/483827173614944526/

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836