cedricp/ddt4all #575: Clio RS EDC, I ruined my radio

Andres86s, 2021-05-18:
Hi all, I bought a thinkdiag obd and wrongly ruined/deleted configurations on my radio, now i have only a screen with the message “unconfigured radio” I bought ddt4all and completely realized that I’m not up to the job, can someone please help?
[image]

testeur70, 2021-05-19:
You bought Ddt4all? The soft is free how did you manage to by a free soft...

AndyIreland, 2021-05-20:
you could try this see if it works: 

Fill out the date (as todays date - the date you programme it) 

Car maker = RS sport (but even Renault will be OK)

touch screen = present

MSM = fitted

[image]

then fill out the VIN and CRC with your details 

then press the home button 5 times - or remove radio fuse inside car leave for 10 mins and put back in and see if its enough filled out to get the radio  working again?

Andres86s, 2021-05-26:
> You bought Ddt4all? The soft is free how did you manage to buy a free soft...

Sorry, perhaps used incorrect word, I obtained ddt4all after the beer donation.

Andres86s, 2021-05-26:
> you could try this see if it works:
>
> Fill out the date (as todays date - the date you programme it)
>
> Car maker = RS sport (but even Renault will be OK)
>
> touch screen = present
>
> MSM = fitted [...]

Thanks so much Andy...unfortunately I think I just made it worse. Now I also have the frightening “add eol settings”. Any recommendations?

AndyIreland, 2021-05-26:
OK - try this , format a 8gb SD Card to fat32 and copy the attached tomtom.000 onto the SD card. then put it into the SD reader of the car - then press the home button 5 times to reboot the R-LINK radio and see if it fixes it. If it does you may have to put in a security code to get the R-Link radio unlocked , you can get the code from your owners manual, or the 'my renault' app or from renault themselves if you give them your VIN number - good luck! 
[https://gofile.io/d/zuOTmd]

Andres86s, 2021-05-27:
> OK - try this , format a 8gb SD Card to fat32 and copy the attached tomtom.000 onto the SD card. then put it into the SD reader of the car - then press the home button 5 times to reboot the R-LINK radio and see if it fixes it. If it does you may have to put in a security code to get the R-Link radio unlocked , you can get the code from your owners manual, or the 'my renault' app or from renault themselves if you give them your VIN number - good luck!
> 
> https://gofile.io/d/zuOTmd
> (Just click 'I HAVE A VPN ALREADY' - if asked)

Thanks Andy, I just did what you've recommended but unfortunately nothing changed. Any other idea or should I pressume I messed up for good?

AndyIreland, 2021-05-27:
> Thanks Andy, I just did what you've recommended but unfortunately nothing changed. Any other idea or should I pressume I messed up for good?

Thats a shame i was really hoping that would help you because I got that Add EOL screen in my Renault Zoe and on restart with the SD card with that TomTom.001 file on the SD card and filling in the VIN and CRC and a few other settings on restart it just asked me to input the security code for the R-LINK and when I did that it all came to life again. 

What about disconnecting the battery and leaving it disconnected for over 10 minutes and reconnecting it? - I suppose you have already tried that? - if you have i am at a loss - your Clio could have a BIC module in it which my Zoe hasnt so maybe when you wiped the configurations maybe it wiped some important detail in the BIC? - its just a guess though at this stage. - if its not that then maybe Renault will have to have it in , take apart the MFD or radio part and find test pins inside on the circuit board and flash the firmware back in that way to it?

Andres86s, 2021-05-28:
> Thats a shame i was really hoping that would help you because I got that Add EOL screen in my Renault Zoe and on restart with the SD card with that TomTom.001 file on the SD card and filling in the [...]

Thanks as always Andy. I actually took it to the Renault dealer and they say that the unit has a short circuit that doesn’t allow them to configure it. Does thar sound reasonable to you? You have to know that in Mexico, dealers are know to be worse than crap so perhaps they are trying to fool me?

AndyIreland, 2021-05-29:
> Thanks as always Andy. I actually took it to the Renault dealer and they say that the unit has a short circuit that doesn’t allow them to configure it. Does thar sound reasonable to you? You have to know that in Mexico, dealers are know to be worse than crap so perhaps they are trying to fool me?

i couldnt say for certain sorry, but i have not heard or read of a short circuit - but I have read somewhere , and i am pretty sure it was the Rlink (not 100% because I read articles on the renault medianav as well) but anyway this article I read said if radio cannot be recovered in other ways they have had to take radio apart and solder a wire to the main board or eeprom and rescue it that way by loading firmware .. I will have a search and see if I can find that article again for you , give me a few days

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif

bigmac5753, 2024-07-01:
where do I find the CRC?

AndyIreland, 2024-07-01:
there is a 'CRC calculator' bult into DDT4ALL on the top menu bar - just put in your car's VIN number and it will give you the CRC

bigmac5753, 2024-07-01:
thanks.

done that and entered them into the boxes but still has same error, unconfigured radio.

AndyIreland, 2024-07-01:
> thanks.
> 
> done that and entered them into the boxes but still has same error, unconfigured radio.

takes a lot more than just writing CRC into box to get unconfigured radio working again.
What did you originally do, press the soft reset button did you?

bigmac5753, 2024-07-01:
yes.

I did everything above and at first it didn't seem to work but i've just gone and tried again and it worked but it had a speaker test tone that I couldn't turn off.  I tried resetting (home x 5) and it now asks for a pin code.

AndyIreland, 2024-07-01:
> yes.
> 
> I did everything above and at first it didn't seem to work but i've just gone and tried again and it worked but it had a speaker test tone that I couldn't turn off. I tried resetting (home x 5) and it now asks for a pin code.

oh well you are nearly there now then - all you have to do is get your radio code from the my renault website after you register your car and add your VIN number - then the test code will be gone

bigmac5753, 2024-07-01:
I have a smart 453 which is basically a twingo 3.  I assume I can't do this through renault?  can't find anywhere on smart to do it.

AndyIreland, 2024-07-01:
> I have a smart 453 which is basically a twingo 3. I assume I can't do this through renault? can't find anywhere on smart to do it.

ah right - I dont know where you would go for that radio code then if that is the car you have but you could look here it might give you pointers:
https://www.evilution.co.uk/?s=rlink