cedricp/ddt4all #576: Renault Trafic mk3 - "no relevant ecu file found): DIAGVERSION [4] SUPPLIER [414] SOFT [C860] VERSION [9070] " - trying to disable start stop system

Maadottaja, 2021-05-22:
Hello all,
I have Renault Trafic (mk3, 2015 model, 1.6dci 120bhp) and I have been using ddt4all successfully to tweak different settings with my van, but I am having headache with one specific ECU, that I want to access.

Upon scanning the ecus for my van with ddt4all, I receive a message: **"( no relevant ecu file found): DIAGVERSION [4] SUPPLIER [414] SOFT [C860] VERSION [9070]"** for the upc-emm module. The ecu database that I use is dated: **9.7.2019** (should be latest), please see on the screenshot below regarding the ecu scan and db.

The 2019 ecu database for my van **"X82 - Trafic3"** includes quite extensive list of upc-emm module SW versions, which seems to contain more SW than other Renault models put together and on that background it seems a bit odd that my older Trafic3 2015 model isnt recognized.

Has anyone else encountered this issue and is there a workaround for it?

**Ultimately what I want to do at the upc-emm ecu is to disable the _"Start and Stop"_ system on the Renault Trafic mk3. Is it safe to use one of the other "upc-emm" ecu SW files as base of adjustment?** 

Thanks in advance for any help and replies.

[image]

KarelSvo, 2021-05-23:
The automatic ecu identification of the database is very neglected by Renault. Select the ecu according to vehicle type and diagnostic version and use the latest ecu file. The Reno CLIP software mostly only identifies using these 2 parameters. 
Your result will be T4_EMM_DDT2000_SW8.5_V1.1_20160112T105701.xml.
with 
<AutoIdents>
<AutoIdent DiagVersion="4" Supplier="414" Soft="C850" Version="9064"/></AutoIdents>

EXcomZkko, 2021-05-25:
> The automatic ecu identification of the database is very neglected by Renault. Select the ecu according to vehicle type and diagnostic version and use the latest ecu file. The Reno CLIP software [...]

Where should we look for these unidentified ECUs? Could you elobrate a little? 

For example; is these pairing is will true? "Soft=C850->SW8.5" Thanks.

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif