cedricp/ddt4all #577: Program versions explanation and backup option

kwaku85, 2021-05-23:
Hello guys!
Few days ago I start to try DDT4all and I have some problems and many questions to You :)

What is the latest version of DDT4all? Full package downloaded from github have daciaklub.pl address instead of program version. I try this one, but it's working very slow. Even first window where I should choose com port is working bad. Should I try to update it using files from github? Or package contain all latest versions?

After that I have try full package from different site. Program says it's 5.5.0 version. How old is it? During installation I noticed it contain older Python (2.7 if i'm correct) and PyQt4 (whatever it is :D) . This one is fast, working much better, Can I use it? Or it's too outdated?

And most important thing- Using this app is little dangerous, especially in newbie hands. Is it possible to read full dump of connected ECUs somehow? Or most importand settings? It should save us from killing cars in case of doing something wrong.

And last question - what about language. Can We somehow translate XML files? Or mayby someone do this already and it can be downloaded from somewhere?

Sorry for so many questions, but I can't find any answers in google, FAQ it's not existist too. And sorry for my english :)

KarelSvo, 2021-05-23:
If you want to save, compare and restore ecu dumps, this software is more suitable. https://gitlab.com/py_ren/pyren/-/blob/master/README.md

kwaku85, 2021-05-23:
You mean pyren in DDT mode? I see it stores dumps during connection, but I can't locate these dumps on HDD. Are these dumps full memory backups or just dumps of settings?

KarelSvo, 2021-05-23:
https://gitlab.com/py_ren/pyren/-/blob/master/pyren/mod_ddt.py#L976

https://gitlab.com/py_ren/pyren/-/blob/master/pyren/mod_ddt.py#L237

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif