cedricp/ddt4all #578: Is DDT4ALL blocked on newer cars or newer cars are behind some kind of firewall?

AndyIreland, 2021-05-26:
Hello - I  have heard a couple of times that DDT4ALL cannot communicate with newer Renaults such as the Clio Mk5 and Zoe ZE50 EV because Renault have blocked the OBD2 socket to allow diagnostics like DDT4ALL or have installed some kind of firewall so end users cannot read or change any parameters - can someone clear this up whether it is true or not? 
Thank you.

KarelSvo, 2021-05-27:
https://github.com/cedricp/ddt4all/issues/404

AndyIreland, 2021-05-27:
Gosh i am glad of this - thank you so much @KarelSvo , i have just written a blog the other day

https://andyrobus.wordpress.com/2021/05/25/changing-infotainment-mmi-settings-on-zoe-ze50-ev-with-ddt4all/

on changing some settings on the A-IVI ECU (I dont own a new renault Zoe ZE50 myself so could not get to try using DDT4ALL on it but I had heard stories about it being 'locked' and only be able to be read and not changed , it was all guesswork and now I know its true

Furtif, 2023-06-02:
https://github.com/cedricp/ddt4all/issues/836

closed 2023-06-02 by Furtif