cedricp/ddt4all #581: copy a key using ddt4all

albfan, 2021-06-09:
Hi,

Is it possible to copy a key using ddt4all

I see this videos: https://www.youtube.com/watch?v=rQrLJenjVHM https://www.youtube.com/watch?v=IgF_lvr6T-I but I'm unsure if something else is needed.

I think I see @cedricp (maintainer) asking questions there, is that you Cedric?

albfan, 2021-06-11:
I see the ddtplugins (card_programming) was removed in migration to python 3) 

This project seems to migrate plugins https://github.com/kreth/ddtplugins/network

Can we reenable plugins?

cedricp, 2021-06-11:
I did not include plugins because a lot of people virginized their ECU thinking it will remove DTCs... And yes that's me asking the question, but I didn't get any reply...

albfan, 2021-06-11:
Hi Cedric.

Makes sense. I'm preparing a PR to reenable them and discuss there.

I was able to copy keys for my car with a ddt4all fork.

Wonder if there's a sniffer option to collect the communications and generalize the process (I'm not sure about the modifications in that fork)

pakoto, 2021-06-11:
what fork do you use?

albfan, 2021-06-11:
RenOlink (I read is a fork somewhere, but looks build on different language) Structure is mostly the same, is it a fork really?

cedricp, 2021-06-11:
Renolink's not a fork, it's a completely separate project. There's a sniffer option in DDT4All, you must select  "can vehicle network" to see what happens on it : 
[image]
let me know about your researches

albfan, 2021-06-14:
Can this sniffer work while other app is using the port?

After reenable card_programming plugin, the first obvious thing is to put a combo box to select ecu file: https://github.com/cedricp/ddtplugins/blob/8c5a787c3ba2c19763317c065904bce880b308fb/card_programming.py#L17

Is there any public roadmap of this card programming plugin? It states clearly it is experimental: https://github.com/cedricp/ddtplugins/blob/8c5a787c3ba2c19763317c065904bce880b308fb/card_programming.py#L127

Any docs out there about the basics on car key learn? From my succesful try I would say it is just car trusting keys and putting some info on keys but have no idea about the whole process.

cedricp, 2021-06-14:
> Can this sniffer work while other app is using the port?
No, but you can use 2 adapters
> After reenable card_programming plugin, the first obvious thing is to put a combo box to select ecu file: https://github.com/cedricp/ddtplugins/blob/8c5a787c3ba2c19763317c065904bce880b308fb/card_programming.py#L17
You could, but this plugin is exclusively dedicated to Megane 2, and won't work on other cars (except Clio 3 AFAIK)
> Is there any public roadmap of this card programming plugin? It states clearly it is experimental: https://github.com/cedricp/ddtplugins/blob/8c5a787c3ba2c19763317c065904bce880b308fb/card_programming.py#L127
No, sorry
> Any docs out there about the basics on car key learn? From my succesful try I would say it is just car trusting keys and putting some info on keys but have no idea about the whole process.
Again, no.

albfan, 2021-06-14:
I'm not sure how to connect two adapters to same odb2 plug in car. Probably snif using wireshark on usb would work then. 

If current plugin is not part of a standard procedure to program keys then I suppose recover all that is meaningless. This needs a plugin from scratch.

Can see how to get progress on this, probably all is about reverse engineering then. Thanks for your replies.

closed 2021-06-14 by albfan